Hyy2001X/AutoBuild-Actions #1: Run Before-Update-Feeds Script 操作时间过长

CurssedCoffin, 2020-09-16:
此项中Checking out luci-app-openclash from https://github.com/vernesong/OpenClash/trunk过多，不知原因为何，尝试编译两次，都是一小时半才能将这一步完成。

closed 2020-09-16 by CurssedCoffin

Hyy2001X, 2020-09-16:
最新版已修复

Hyy2001X, 2020-09-16:
日志
2020-09-16T11:22:01.1914535Z package/Makefile:65: recipe for target 'package/install' failed
2020-09-16T11:22:01.1915347Z make[2]: Leaving directory '/home/runner/work/AutoBuild-Actions/AutoBuild-Actions/openwrt'
2020-09-16T11:22:01.1916583Z package/Makefile:108: recipe for target '/home/runner/work/AutoBuild-Actions/AutoBuild-Actions/openwrt/staging_dir/target-mipsel_24kc_musl/stamp/.package_install' failed
2020-09-16T11:22:01.1920082Z make[1]: *** [/home/runner/work/AutoBuild-Actions/AutoBuild-Actions/openwrt/staging_dir/target-mipsel_24kc_musl/stamp/.package_install] Error 2
2020-09-16T11:22:01.1921033Z Build failed - please re-run with -j1 to see the real error message
2020-09-16T11:22:01.1921403Z make: *** [world] Error 1
2020-09-16T11:22:01.1922104Z make[1]: Leaving directory '/home/runner/work/AutoBuild-Actions/AutoBuild-Actions/openwrt'
2020-09-16T11:22:01.1923151Z /home/runner/work/AutoBuild-Actions/AutoBuild-Actions/openwrt/include/toplevel.mk:222: recipe for target 'world' failed
2020-09-16T11:22:01.1938858Z ##[error]Process completed with exit code 2.