Hyy2001X/AutoBuild-Actions #2: 你的CONFIG不应该这样写

z826540272, 2020-09-17:
这样写出来的config文件后期源码更新之后可能伴随不兼容或镜像体积过大的问题。
你在menuconfig里改了什么就写什么。
比如改了设备成D2,就只用写这三行，自行查询config.seed用法。
CONFIG_TARGET_ramips=y
CONFIG_TARGET_ramips_mt7621=y
CONFIG_TARGET_ramips_mt7621_DEVICE_d-team_newifi-d2=y

https://github.com/z826540272/passwall-openwrt/blob/master/newifi3.config
这是我的NEWIFI3配置。

Hyy2001X, 2020-09-18:
偷懒了这波，直接用了menuconfig生成的config

CurssedCoffin, 2020-09-18:
但是直接修改config文件不是会造成无法编译吗

z826540272, 2020-09-19:
不会 发送自 Windows 10 版邮件应用 发件人: CurssedCoffin发送时间: 2020年9月19日 3:29收件人: Hyy2001X/AutoBuild-Actions抄送: z826540272; Author主题: Re: [Hyy2001X/AutoBuild-Actions] 你的CONFIG不应该这样写 (#2) 但是直接修改config文件不是会造成无法编译吗

CurssedCoffin, 2020-09-19:
请问如何生成这种简短的.config文件

Hyy2001X, 2020-09-19:
/bin/targets/ramips/mt7621/config.seed

z826540272, 2020-09-19:
先生成config.seed,再手动删除自动添加的，保留手动修改的部分 发送自 Windows 10 版邮件应用 发件人: CurssedCoffin发送时间: 2020年9月19日 21:04收件人: Hyy2001X/AutoBuild-Actions抄送: z826540272; Author主题: Re: [Hyy2001X/AutoBuild-Actions] 你的CONFIG不应该这样写 (#2) 请问如何生成这种简短的.config文件

CurssedCoffin, 2020-09-19:
> /bin/targets/ramips/mt7621/config.seed

需要编译过一次固件吗，我的机器没编译过，没有bin文件夹

CurssedCoffin, 2020-09-19:
请问您可以写一个客制化固件，从别的源添加包的教程吗？谢谢您！

Hyy2001X, 2020-09-20:
> > /bin/targets/ramips/mt7621/config.seed
> 
> 需要编译过一次固件吗，我的机器没编译过，没有bin文件夹

make menuconfig一次之后在终端输入 .scripts/diffconfig.sh > defconfig,执行完成后在源码根目录查看defconfig文件

closed 2020-12-03 by Hyy2001X