Hyy2001X/AutoBuild-Actions #4: 添加插件

CurssedCoffin, 2020-09-19:
请问可以添加ssr plus的 ssr server服务吗，此版默认只带socks5，对于一些有公网ip的用户ssr server很有用，我尝试直接在config文件中添加但是编译报错
另外请问您本地生成config的环境可以大概描述一下吗，目前我用的是lean的仓库外加一些额外的包

Hyy2001X, 2020-09-20:
.config删除#CONFIG_PACKAGE_luci-app-ssr-plus_INCLUDE_ShadowsocksR_Server is not set
再添加CONFIG_PACKAGE_luci-app-ssr-plus_INCLUDE_ShadowsocksR_Server=y

先添加serverchan源码，然后在.config添加CONFIG_PACKAGE_luci-app-serverchan=y

Hyy2001X, 2020-09-20:
添加额外的包方法 
1. fork AutoBuild-Actions最新仓库
2. 编辑Scripts/diy-script.sh
3. 找到Diy-Part1() {
}
4. 添加luci-app-serverchan示例：
ExtraPackages git luci-app-serverchan https://github.com/tty228 master

CurssedCoffin, 2020-09-20:
谢谢！

CurssedCoffin, 2020-09-20:
老大，我又来问问题了。。
向workflow里添加了UPLOAD_BIN_DIR: true字段，并没有上传整个bin文件夹，而且添加release报错
[image]
P.S. 我加了几个包编译但不编译到固件里，需要从bin中提取IPK。。。，因为新3的内存太小了，我需要刷固件后设置U盘为overlay然后再添加这几个包

Hyy2001X, 2020-09-20:
需要先生成token，方法参考https://blog.csdn.net/sytandxly/article/details/81100010
然后导入TOKEN到你的AutoBuild-Actions项目里，方法如下
你的AutoBuild-Actions 点右边Settings，然后点Secrets，New Secrets，Name填写RELEASE_TOKEN，Value填你刚获取到的token

Hyy2001X, 2020-09-20:
上传整个bin文件夹已经被我取消了，需要自己编辑workflow文件

CurssedCoffin, 2020-09-20:
[image]
编辑了workflow文件并添加了UPLOAD_BIN_DIR: true字段，action中只上传了固件。。。

Hyy2001X, 2020-09-20:
UPLOAD_BIN_DIR: true只是赋值 调用部分需要自己添加
    - name: Upload bin directory
      uses: actions/upload-artifact@main
      if: env.UPLOAD_BIN_DIR == 'true' && !cancelled()
      with:
        name: OpenWrt_bin
        path: openwrt/bin
这段添加到Upload Firmware to Artifacts 与 Generate Tag的中间

CurssedCoffin, 2020-09-20:
谢谢！

Hyy2001X, 2020-09-20:
需要把Customize/AutoUpdate.sh中第 6 10 行改为自己的信息

CurssedCoffin, 2020-09-20:
请问在添加一些内核方面的包比如nginx和php时，会报错Package size mismatch，这是源的问题吗？ 尝试更换为官方源依旧会报错。
[image]

Hyy2001X, 2020-09-20:
> 请问在添加一些内核方面的包比如nginx和php时，会报错Package size mismatch，这是源的问题吗？ 尝试更换为官方源依旧会报错。
> [image]

试试opkg install nginx --force-checksum

CurssedCoffin, 2020-09-20:
老哥，你的新workflow是不是，编译不了，我特意跑到你的仓库跑一个action编译，都编译不了。。。报错

CurssedCoffin, 2020-09-22:
这咋有的时候会编译报错退出啊老哥，我的和你的我都跑了一次，都报错，是gayhub的问题吗，没头绪。。。

CurssedCoffin, 2020-09-22:
上了一天学，刚爬到电脑上看了下log，发现是frp报错，估计是上游的问题，版本0.34 curl失败，0.33正常，再尝试编译下

closed 2020-09-22 by CurssedCoffin

CurssedCoffin, 2020-09-26:
老哥，提个建议哈，配置文件里加一个kmod-tun，不然openclash的tun模式无法正常使用