Hyy2001X/AutoBuild-Actions #7: 运行一段时间后（一般是30小时）进不去luci

CurssedCoffin, 2020-10-12:
[image]
如图，运行超过一天左右便会找不到luci，必须手动重启才能正常显示，目前用每天自动重启的方式解决
运行的服务有：ssrp ssrp服务端 openclash
在外置usb运行系统，服务启动正常，只有luci挂掉
原因不明。。。
求解答

Hyy2001X, 2020-10-12:
最近没时间折腾了，项目停一段时间。。

CurssedCoffin, 2020-10-12:
好的，放松享受生活！！！

closed 2020-10-12 by CurssedCoffin

CurssedCoffin, 2020-10-17:
问题好像更严重了，之前24h自动重启没有见过 luci 页面访问不了的情况，现在是24小时内就会有一次
下面是刚刚重启，上一次运行的系统log
```
Sat Oct 17 15:54:47 2020 kern.debug kernel: [    0.000000] On node 0 totalpages: 131072
Sat Oct 17 15:54:47 2020 kern.debug kernel: [    0.000000]   Normal zone: 576 pages used for memmap
Sat Oct 17 15:54:47 2020 kern.debug kernel: [    0.000000]   Normal zone: 0 pages reserved
Sat Oct 17 15:54:47 2020 kern.debug kernel: [    0.000000]   Normal zone: 65536 pages, LIFO batch:15
Sat Oct 17 15:54:47 2020 kern.debug kernel: [    0.000000]   HighMem zone: 65536 pages, LIFO batch:15
Sat Oct 17 15:54:47 2020 kern.info kernel: [    0.000000] percpu: Embedded 14 pages/cpu s27184 r8192 d21968 u57344
Sat Oct 17 15:54:47 2020 kern.debug kernel: [    0.000000] pcpu-alloc: s27184 r8192 d21968 u57344 alloc=14*4096
Sat Oct 17 15:54:47 2020 kern.debug kernel: [    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
Sat Oct 17 15:54:47 2020 kern.info kernel: [    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 130496
Sat Oct 17 15:54:47 2020 kern.notice kernel: [    0.000000] Kernel command line: console=ttyS0,115200 rootfstype=squashfs,jffs2
Sat Oct 17 15:54:47 2020 kern.info kernel: [    0.000000] Dentry cache hash table entries: 32768 (order: 5, 131072 bytes, linear)
Sat Oct 17 15:54:47 2020 kern.info kernel: [    0.000000] Inode-cache hash table entries: 16384 (order: 4, 65536 bytes, linear)
Sat Oct 17 15:54:47 2020 kern.info kernel: [    0.000000] Writing ErrCtl register=00068824
Sat Oct 17 15:54:47 2020 kern.info kernel: [    0.000000] Readback ErrCtl register=00068824
Sat Oct 17 15:54:47 2020 kern.info kernel: [    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
Sat Oct 17 15:54:47 2020 kern.info kernel: [    0.000000] Memory: 509860K/524288K available (5737K kernel code, 250K rwdata, 1240K rodata, 1272K init, 238K bss, 14428K reserved, 0K cma-reserved, 262144K highmem)
Sat Oct 17 15:54:47 2020 kern.info kernel: [    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
Sat Oct 17 15:54:47 2020 kern.info kernel: [    0.000000] rcu: Hierarchical RCU implementation.
Sat Oct 17 15:54:47 2020 kern.info kernel: [    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 25 jiffies.
Sat Oct 17 15:54:47 2020 kern.info kernel: [    0.000000] NR_IRQS: 256
Sat Oct 17 15:54:47 2020 kern.notice kernel: [    0.000000] random: get_random_bytes called from start_kernel+0x340/0x550 with crng_init=0
Sat Oct 17 15:54:47 2020 kern.info kernel: [    0.000000] CPU Clock: 880MHz
Sat Oct 17 15:54:47 2020 kern.info kernel: [    0.000000] clocksource: GIC: mask: 0xffffffffffffffff max_cycles: 0xcaf478abb4, max_idle_ns: 440795247997 ns
Sat Oct 17 15:54:47 2020 kern.info kernel: [    0.000000] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 4343773742 ns
Sat Oct 17 15:54:47 2020 kern.info kernel: [    0.000010] sched_clock: 32 bits at 440MHz, resolution 2ns, wraps every 4880645118ns
Sat Oct 17 15:54:47 2020 kern.info kernel: [    0.007810] Calibrating delay loop... 583.68 BogoMIPS (lpj=1167360)
Sat Oct 17 15:54:47 2020 kern.info kernel: [    0.042005] pid_max: default: 32768 minimum: 301
Sat Oct 17 15:54:47 2020 kern.info kernel: [    0.046781] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
Sat Oct 17 15:54:47 2020 kern.info kernel: [    0.054007] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
Sat Oct 17 15:54:47 2020 kern.info kernel: [    0.064561] rcu: Hierarchical SRCU implementation.
Sat Oct 17 15:54:47 2020 kern.info kernel: [    0.070037] smp: Bringing up secondary CPUs ...
Sat Oct 17 15:54:47 2020 kern.warn kernel: [    0.672123] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
Sat Oct 17 15:54:47 2020 kern.warn kernel: [    0.672132] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
Sat Oct 17 15:54:47 2020 kern.warn kernel: [    0.672144] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
Sat Oct 17 15:54:47 2020 kern.info kernel: [    0.672252] CPU1 revision is: 0001992f (MIPS 1004Kc)
Sat Oct 17 15:54:47 2020 kern.info kernel: [    0.102802] Synchronize counters for CPU 1: done.
Sat Oct 17 15:54:47 2020 kern.warn kernel: [    4.633201] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
Sat Oct 17 15:54:47 2020 kern.warn kernel: [    4.633211] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
Sat Oct 17 15:54:47 2020 kern.warn kernel: [    4.633219] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
Sat Oct 17 15:54:47 2020 kern.info kernel: [    4.633281] CPU2 revision is: 0001992f (MIPS 1004Kc)
Sat Oct 17 15:54:47 2020 kern.info kernel: [    0.167546] Synchronize counters for CPU 2: done.
Sat Oct 17 15:54:47 2020 kern.warn kernel: [    8.379857] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
Sat Oct 17 15:54:47 2020 kern.warn kernel: [    8.379865] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
Sat Oct 17 15:54:47 2020 kern.warn kernel: [    8.379874] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
Sat Oct 17 15:54:47 2020 kern.info kernel: [    8.379936] CPU3 revision is: 0001992f (MIPS 1004Kc)
Sat Oct 17 15:54:47 2020 kern.info kernel: [    0.225412] Synchronize counters for CPU 3: done.
Sat Oct 17 15:54:47 2020 kern.info kernel: [    0.255292] smp: Brought up 1 node, 4 CPUs
[... cut ...]
Sat Oct 17 15:55:06 2020 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Sat Oct 17 15:55:07 2020 cron.err crond[3543]: crond (busybox 1.31.1) started, log level 8
Sat Oct 17 15:55:07 2020 daemon.crit dnsmasq[3505]: cannot read /tmp/dnsmasq.ssr/gfw_list.confjHcNGD: No such file or directory
Sat Oct 17 15:55:07 2020 daemon.crit dnsmasq[3505]: FAILED to start up
Sat Oct 17 15:55:07 2020 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Sat Oct 17 15:55:07 2020 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
```