Hyy2001X/AutoBuild-Actions #33: 能不能增加内网穿透NPS服务端的插件

sirliu, 2021-01-28:
能不能增加内网穿透NPS服务端的插件

NaiHeKK, 2021-01-28:
luci-app-nps我发现package里有个这个包

sirliu, 2021-01-28:
> luci-app-nps

你的源能分享下嘛，我的是这样的源，里面没有发现，设备是newifi3 d2：
`src/gz openwrt_core https://mirrors.cloud.tencent.com/lede/snapshots/targets/ramips/mt7621/packages
src/gz openwrt_base https://mirrors.cloud.tencent.com/lede/snapshots/packages/mipsel_24kc/base
src/gz openwrt_freifunk https://mirrors.cloud.tencent.com/lede/snapshots/packages/mipsel_24kc/freifunk
src/gz openwrt_luci https://mirrors.cloud.tencent.com/lede/releases/18.06.8/packages/mipsel_24kc/luci
src/gz openwrt_packages https://mirrors.cloud.tencent.com/lede/snapshots/packages/mipsel_24kc/packages
src/gz openwrt_routing https://mirrors.cloud.tencent.com/lede/snapshots/packages/mipsel_24kc/routing
src/gz openwrt_telephony https://mirrors.cloud.tencent.com/lede/snapshots/packages/mipsel_24kc/telephony`

NaiHeKK, 2021-01-28:
> 你的源能分享下嘛，我的是这样的源，里面没有发现，设备是newifi3 d2：
> `src/gz openwrt_core https://mirrors.cloud.tencent.com/lede/snapshots/targets/ramips/mt7621/packages src/gz openwrt_base [...]

直接在config里添加就行应该

sirliu, 2021-01-28:
> 直接在config里添加就行应该

已经测试，那只是客户端……

Hyy2001X, 2021-01-30:
不能开机自启动试试设置corntab

sirliu, 2021-01-31:
> 不能开机自启动试试设置corntab

OK，我试一试吧

closed 2021-02-04 by sirliu