Hyy2001X/AutoBuild-Actions #36: 添加bypass和helloworld后编译失败

NaiHeKK, 2021-02-12:
你好，我在【AutoBuild_DiyScript.sh】里打开了bypass和helloworld并且在【NEWIFI_D2-NIGHTLY.config】里添加了
`CONFIG_PACKAGE_luci-app-bypass=y`
`CONFIG_PACKAGE_luci-app-vssr=y`
但是编译的时候会报错
`make -r world: build failed. Please re-run make with -j1 V=s or V=sc for a higher verbosity level to see what's going on
/home/runner/work/AutoBuild-Actions/AutoBuild-Actions/openwrt/include/toplevel.mk:222: recipe for target 'world' failed
make: *** [world] Error 1
WARNING: Makefile 'package/other/luci-app-bypass/Makefile' has a dependency on 'lua-maxminddb', which does not exist
WARNING: Makefile 'package/lean/luci-app-vssr/Makefile' has a dependency on 'lua-maxminddb', which does not exist`

[日志](https://github.com/NaiHeKK/AutoBuild-Actions/runs/1888365907?check_suite_focus=true)

请问这是哪里配错了吗？
这是我的[仓库](https://github.com/NaiHeKK/AutoBuild-Actions)

NaiHeKK, 2021-02-14:
就好奇怪啊，现在就多开了个passwall都编译不过了

Hyy2001X, 2021-02-14:
bypass helloworld ssrplus+ passwall 这几个插件同时安装可能会有冲突，建议只选其中一个
passwall和ssrplus+可以完美共存，其他的没测试过

NaiHeKK, 2021-02-14:
> bypass helloworld ssrplus+ passwall 这几个插件同时安装可能会有冲突，建议只选其中一个
> passwall和ssrplus+可以完美共存，其他的没测试过

passwall和ssrplue+以前可以共存，但是从前几天开始就不行了，我都没动配置，前几天突然就编译失败了。重新fork你的仓库只开启passwall都会导致编译失败

Hyy2001X, 2021-02-15:
回滚一下mt76的版本试试

NaiHeKK, 2021-02-16:
> 回滚一下mt76的版本试试

试了一下，passwall可以了
helloworld估计和ssrplus+冲突，编译不过

NaiHeKK, 2021-02-16:
helloworld这些到底哪个好用点啊，这么多

closed 2021-02-22 by NaiHeKK