Hyy2001X/AutoBuild-Actions #41: 关于config和AutoBuild DiyScript软件包对应关系的疑问

NaiHeKK, 2021-02-24:
以HelloWorld为例：
> 在AutoBuild DiyScript里为INCLUDE_HelloWorld
> 在AutoBuild Functions里看起来包名为luci-app-vssr
>  那么在config里应该怎么写

看passwall的话：
> 在AutoBuild DiyScript里为INCLUDE_Passwall
> 在AutoBuild Functions里看起来包名为OpenClash
> 在config里却是CONFIG_PACKAGE_luci-app-passwall

求大佬解答一下：
> 1. 这之间咋对应的啊？

> 2. config应该根据什么来写啊？

Hyy2001X, 2021-02-24:
这其中没有什么对应关系
例如 INCLUDE_HelloWorld，它只是一个开关(名字是我自己取的)，其作用就只是当它为 true 时，脚本会自动帮你下载一个名叫 luci-app-vssr 的软件包到源码的目录里面，方便有些需要这些软件包的人不用再在Diy-Script.sh 中使用 git svn 这类的代码去添加这个包，只需要修改一下 .config 就可以了

config就根据软件包的真实包名来写，跟 INCLUDE_XXX 没有关系

因为怕有些人因为代码太复杂搞混我才把代码分离成 DiyScript.sh 和 Function.sh 两个文件，一般你们只需要编辑DiyScript.sh就够了

Hyy2001X, 2021-02-24:
可以看一下Readme

NaiHeKK, 2021-02-25:
> Readme

好的，谢谢了

closed 2021-02-25 by NaiHeKK