Hyy2001X/AutoBuild-Actions #49: 关于添加PassWall

NaiHeKK, 2021-03-28:
大佬，现在要添加**Passwall**：`ExtraPackages git other openwrt-passwall https://github.com/xiaorouji main`这个应该放在`Scripts/AutoBuild_DiyScript.sh`的Diy-Part2里还是`CustomPackages/d-team_newifi-d2`里面啊？

Hyy2001X, 2021-03-28:
都可，专门添加一个CustomPackages是为了多设备的编译。
想添加到Diy-Part2里就写ExtraPackages git other openwrt-passwall https://github.com/xiaorouji main
想添加CustomPackages到里就写git other openwrt-passwall https://github.com/xiaorouji main

NaiHeKK, 2021-03-30:
> 都可，专门添加一个CustomPackages是为了多设备的编译。
> 想添加到Diy-Part2里就写ExtraPackages git other openwrt-passwall https://github.com/xiaorouji main
> 想添加CustomPackages到里就写git other openwrt-passwall https://github.com/xiaorouji main

好的，谢谢大佬

closed 2021-03-30 by NaiHeKK