Hyy2001X/AutoBuild-Actions #51: 提几点小建议

teasiu, 2021-04-04:
首先非常感谢作者的autoupdate脚本，这个非常棒。
另外提几点小建议：
1。针对x86_64，firmware的size建议200M+500M,因为coolshare的也是这个，
有些玩家从coolshare开始转移过来，可以避免重新格sda3的位置。
2。你的脚本只能依赖github来编译，如何从本地可以编译呢？
既能本地编译，又能github action，并且相互搭配，制作autoupdate.
我水平有限，瞎改了一些代码，在https://github.com/teasiu/lede-x86_64
请参考。我实现了简单的互相搭配。限于水平不够，还请作者参考并实现完善到这里。
3。lede里面的自带autosamba和京东签到是很多余的，我用patch实现了自动修正。
4。关于实现原版19.07或21.02，是可以的，
请参考https://github.com/slitaz-cn/dragino2-19.07.7
增加一个scripts/convert_translation.sh脚本就可以实现和18.06的app转换。

再次感谢作者的无私奉献。

神雕

teasiu, 2021-04-04:
另外：
作者的action里面的release upload的那个key是单独的，最好修正为全局的，
否着很多网友fork了以后，auto action是失败的。
GITHUB_TOKEN: ${{ secrets.GITHUB_TOKEN }}

Hyy2001X, 2021-04-04:
非常感谢你的建议和反馈，部分提议已经修改并提交
本地编译脚本在做了

teasiu, 2021-04-04:
感谢作者的回应。
另外补充几个小地方：
1。全局的key不需要另行建立，用户只要网页能登陆，${{ secrets.GITHUB_TOKEN }}就存在了。所以你的readme里面写上反而让用户不知所措。当然，进阶用户自然会跳过这个说明。
2。你的luci-app-autoupdate的checkupdate.sh里面，那个tmp剩余空间检测的30M，在我其中一个64M缓存的路由器上怎么都不够，建议调整到25-27之间。
3。还是autoupdate，一般国内用户都难于直通github，能不能将检测和下载地址更换为“加速地址”呢？
比如参考这个https://github.com/teasiu/hg255d/releases/latest
也可以参考这个https://blog.csdn.net/networken/article/details/105122778

再次感谢作者的采纳。
神雕

Hyy2001X, 2021-04-05:
> 感谢作者的回应。
> 另外补充几个小地方：
> 1。全局的key不需要另行建立，用户只要网页能登陆，${{ secrets.GITHUB_TOKEN }}就存在了。所以你的readme里面写上反而让用户不知所措。当然，进阶用户自然会跳过这个说明。 [...]

AutoUpdate.sh V5.5 已调整相关策略，当前固件剩余空间强制检测仅对 x86 设备生效,非 x86 设备调整为自动检测云端固件大小并与剩余空间进行比较，通过后即可下载更新固件。

Hyy2001X, 2021-04-05:
> 感谢作者的回应。
> 另外补充几个小地方：
> 1。全局的key不需要另行建立，用户只要网页能登陆，${{ secrets.GITHUB_TOKEN }}就存在了。所以你的readme里面写上反而让用户不知所措。当然，进阶用户自然会跳过这个说明。 [...]

已添加 FastGit 镜像站支持,当 Google检测失败时会自动启用镜像加速下载固件,检测版本还是默认用 Github

teasiu, 2021-04-05:
加速下载太棒了，我测试了，秒下。

Hyy2001X, 2021-05-19:
关于第一点修改firmware size,这样会导致使用旧版固件的更新到新版以后因为区表不一致从而丢失所有数据,x86的设备肯定不止overlay底下的那些文件,备份-恢复数据太麻烦

teasiu, 2021-05-19:
> 关于第一点修改firmware size,这样会导致使用旧版固件的更新到新版以后因为区表不一致从而丢失所有数据,x86的设备肯定不止overlay底下的那些文件,备份-恢复数据太麻烦

good job for all!

加我微信teasiu
有很多东西交流一下。

closed 2021-05-26 by Hyy2001X