Hyy2001X/AutoBuild-Actions #53: 原生openwrt19.07和21.02添加ssr和passwall的脚本补丁

teasiu, 2021-04-06:
关于原生openwrt添加ssr和passwall，很多网友编译失败，是因为缺少了upx和ucl.
这里将方法写出来，供作者参考利用。
手动添加：（原生openwrt源码）
1.将lean的lede下tools/upx和tools/ucl拷贝到 原生openwrt源码tools下
2.修改tools/Makefile,（33行和46行附近）
-tools-y += lzma squashfskit4 zip
+tools-y += lzma squashfskit4 zip ucl upx
+$(curdir)/upx/compile := $(curdir)/ucl/compile
自动添加：（打补丁）
19.07.7的补丁：
https://raw.githubusercontent.com/slitaz-cn/dragino2-19.07.7/main/patches/0003-add-upx-ucl-for-19.07.7.patch
21.02的补丁：
https://github.com/fanck0605/openwrt-nanopi-r2s/raw/openwrt-21.02/patches/0004-tools-add-ucl-and-upx-compresss.patch

缺少上面的补丁，原生openwrt编译ssrp和passwall会失败。
一定要打上。
配合scripts/convert_translation.sh
就可以把大量的18.06的app转过来了。

希望能帮到大家少走弯路。

神雕

teasiu, 2021-04-06:
打补丁方法：
下载0003-add-upx-ucl-for-19.07.7.patch补丁
cd openwrt
patch < 0003-add-upx-ucl-for-19.07.7.patch -p1
或者
cat 0003-add-upx-ucl-for-19.07.7.patch | patch -p1

Hyy2001X, 2021-04-06:
收到，等我回去搞一搞😁

teasiu, 2021-04-06:
忘了说还要替换golang
cd openwrt
rm -rf feeds/packages/lang/golang
svn co https://github.com/coolsnowwolf/packages/trunk/lang/golang feeds/packages/lang/golang

详情看这里https://github.com/slitaz-cn/dragino2-19.07.7/blob/main/build.sh

Hyy2001X, 2021-04-06:
感谢你的建议，改动已提交(未进行完整测试)

teasiu, 2021-04-06:
重大问题，你的scripts在开始的replace或extract时，把package文件夹下的文件全部删了，只剩下other lean 
请看记录https://github.com/teasiu/openwrt-autoupdate/runs/2280176066?check_suite_focus=true

teasiu, 2021-04-06:
后来我ssh进去，用git restore恢复回去，才能继续进行测试。

teasiu, 2021-04-06:
[00:55:35] Loading Custom Packages list: [x86_64] ...
[00:55:35] Checking out package [rtl8821cu] to package/kernel ...
[00:55:37] Package [rtl8821cu] is detected!
[00:55:37] Removing old package [] ...
[00:55:37] Checking out package [] to package/ ...
[00:55:37] Error option:  !

teasiu, 2021-04-06:
[02:14:31] Creating new folder [bin/Firmware] ...
Firmware: AutoBuild--20210407.bin
mv: cannot stat 'bin/targets/x86/64/openwrt-x86-64--squashfs-sysupgrade.bin': No such file or directory
Error: Process completed with exit code 1.
当源码是openwrt-19.07时，$version未设置，类似lede的R21.3.27未设置。
https://github.com/teasiu/openwrt-autoupdate/runs/2280268812?check_suite_focus=true

teasiu, 2021-04-07:
我看明白了，原生openwrt要添加leagcy和efi的output才能适应你的脚本。我研究一下。

Hyy2001X, 2021-04-07:
> 重大问题，你的scripts在开始的replace或extract时，把package文件夹下的文件全部删了，只剩下other lean
> 请看记录https://github.com/teasiu/openwrt-autoupdate/runs/2280176066?check_suite_focus=true

应该修好了

Hyy2001X, 2021-04-07:
> [02:14:31] Creating new folder [bin/Firmware] ...
> Firmware: AutoBuild--20210407.bin
> mv: cannot stat 'bin/targets/x86/64/openwrt-x86-64--squashfs-sysupgrade.bin': No such file or directory
> Error: [...]

不太建议修改 AutoBuild_Function.sh 这个文件，我测试编译19.07的时候没有出现这个问题

teasiu, 2021-04-07:
我再测试看看，另外，需要再添加一个补丁，就是默认使用dnsmasq-full，否则编译ssrp会出错。
https://github.com/teasiu/openwrt-autoupdate/blob/main/CustomFiles/Patches/0004-set-default-dnsmasq-full.patch

teasiu, 2021-04-07:
测试结果如下：
1.在AutoBuild_DiyScript.sh添加Default_Device=x86_64
编译能完成，但是release为空
https://github.com/teasiu/openwrt-autoupdate/actions/runs/725750122
2.AutoBuild_DiyScript.sh添加Default_Device= （默认留空）
编译无法完成，
显示
[Preload] All done !
[20:09:34] Creating new folder [bin/Firmware] ...
Firmware: AutoBuild--R19.07-20210407.bin
mv: cannot stat 'bin/targets/x86/64/openwrt-x86-64--squashfs-sysupgrade.bin': No such file or directory
Error: Process completed with exit code 1.

Hyy2001X, 2021-04-07:
> 测试结果如下：
> 1.在AutoBuild_DiyScript.sh添加Default_Device=x86_64
> 19.07编译能完成，但是release为空
> https://github.com/teasiu/openwrt-autoupdate/actions/runs/725750122
> 2.AutoBuild_DiyScript.sh添加Default_Device= （默认留空）
> 编译无法完成，
> 显示
> [Preload] All done !
> [20:09:34] Creating new folder [bin/Firmware] ...
> Firmware: AutoBuild--R19.07-20210407.bin
> mv: cannot stat 'bin/targets/x86/64/openwrt-x86-64--squashfs-sysupgrade.bin': No such file or directory
> Error: Process completed with exit code 1.
> 3.在AutoBuild_DiyScript.sh添加Default_Device=x86_64
> 21.02编译能完成，release完成。这个ok。

我都测试一下，找找问题所在

teasiu, 2021-04-08:
今天的修改，我继续测试了。结果和昨天一样。19.07的release出错。21.02则OK。
出错原因显示脚本判断为bin后缀去重命名，然后当然找不到文件了。

teasiu, 2021-04-08:
Run cd openwrt 

18config.buildinfo 

19feeds.buildinfo 

20openwrt-x86-64-combined-ext4.img.gz 

21openwrt-x86-64-combined-squashfs.img.gz 

22openwrt-x86-64-generic-rootfs.tar.gz 

23openwrt-x86-64-generic.manifest 

24openwrt-x86-64-rootfs-ext4.img.gz 

25openwrt-x86-64-rootfs-squashfs.img.gz 

26openwrt-x86-64-vmlinuz 

27packages 

28sha256sums 

29version.buildinfo 

30[21:53:35] Creating new folder [bin/Firmware] ... 

31[Preload Info] Default_Firmware: openwrt-x86-64--squashfs-sysupgrade.bin 

32[Preload Info] AutoBuild_Firmware: AutoBuild--R19.07-20210408.bin 

33Error: 35] [ERROR] Common Firmware is not detected ! 

34[21:53:35] Actions Avaliable: 41G

Hyy2001X, 2021-04-08:
> Run cd openwrt
>
> 18config.buildinfo
>
> 19feeds.buildinfo
>
> 20openwrt-x86-64-combined-ext4.img.gz
>
> 21openwrt-x86-64-combined-squashfs.img.gz
>
> 22openwrt-x86-64-generic-rootfs.tar.gz [...]

这两天有点事情，这个得推到下周了

teasiu, 2021-04-08:
我找到原因了：
openwrt-19.07的config模板里对CONFIG_TARGET_x86_64_DEVICE_generic这句有差异。
它的唯一体现是
CONFIG_TARGET_x86_64_generic=y
而不是
CONFIG_TARGET_x86_64_DEVICE_generic=y

所以，你的脚本提取字符时，第33行
egrep -o "CONFIG_TARGET.*DEVICE.*=y" .config是不成功的。
应该修改？或
egrep -o "CONFIG_TARGET.*generic.*=y" .config  ?（为了不改变其他的config变量，可能还需要条件判断）

结果就是造成变量TARGET_PROFILE不是正确的x86_64
后面的选择就自动跳转变量为bin了。
所以最后release就反应不出来了。

teasiu, 2021-04-08:
我改成这样：
	x86_Test="$(egrep -o "CONFIG_TARGET.*DEVICE.*=y" .config | sed -r 's/CONFIG_TARGET_(.*)_DEVICE_(.*)=y/\1/')"
	x86_Test2="$(egrep -o "CONFIG_TARGET.*generic.*=y" .config | sed -r 's/CONFIG_TARGET_(.*)_generic=y/\1/')"
	if [[ "${x86_Test}" == "x86_64" ]] || [[ ${x86_Test2}" == "x86_64 ]];then
		TARGET_PROFILE="x86_64"
貌似解决了。

teasiu, 2021-04-08:
继续output固件名称错误：
实际固件名称openwrt-x86-64-combined-squashfs.img.gz
脚本拟定名称openwrt-x86-64-generic-squashfs-combined.img.gz
我作了如下修正：（约172行）
	case ${Current_Branch} in
		19.07)
			_Firmware=openwrt
			_Legacy_Firmware=combined-squashfs
			_EFI_Firmware=combined-squashfs-efi
		;;
		*)
			_Firmware=openwrt
			_Legacy_Firmware=generic-squashfs-combined
			_EFI_Firmware=generic-squashfs-combined-efi
		;;
		esac
貌似搞定了。

Hyy2001X, 2021-04-09:
> 继续output固件名称错误：
> 实际固件名称openwrt-x86-64-combined-squashfs.img.gz
> 脚本拟定名称openwrt-x86-64-generic-squashfs-combined.img.gz
> 我作了如下修正：（约172行）
> case ${Current_Branch} in
> 19.07)
> _Firmware=openwrt
> _Legacy_Firmware=combined-squashfs
> _EFI_Firmware=combined-squashfs-efi
> ;;
> *)
> _Firmware=openwrt
> _Legacy_Firmware=generic-squashfs-combined
> _EFI_Firmware=generic-squashfs-combined-efi
> ;;
> esac
> 貌似失败了
> 无奈之下，我又瞎改了182行
> if [ -f openwrt-x86-64-combined-squashfs.img.gz ];then
> mv -f openwrt-x86-64-combined-squashfs.img.gz openwrt-x86-64-generic-squashfs-combined.img.gz
> echo "调整了文件名openwrt-x86-64-generic-squashfs-combined.img.gz"
> fi
> 强迫该文件名以适应脚本。

这个19.07这么奇葩，是我没有想到的，现在正在测试各种源码和分支的兼容性

teasiu, 2021-04-09:
ok,瞎改以后，19.07的固件终于出来了。
https://github.com/teasiu/openwrt-autoupdate/releases/tag/AutoUpdate

接下来我就作config调整适用编译。
有很多补丁和app源码要调整。

teasiu, 2021-04-09:
以下几点建议期望完善：
1.workflows下，增加x86_64-19.07.yml和x86_64-21.02.yml
2.scripts/AutoBuild_Function.sh的补丁脚本修改为自主添加多个patch
类似：（以星号代表任意？）
Replace_File CustomFiles/Patches/*-${Current_Branch}.patch ./
cat *-${Current_Branch}.patch | patch -p1 > /dev/null 2>&1
3.Auto_ExtraPackages()的部分，添加x86_64-19.07和x86_64-21.02的识别，
文件夹CustomPackages内添加上面两个文档
4.Configs文件夹增加x86_64-19.07-config.seed和x86_64-21.02-config.seed
并反应到workflows的相应yml中。

Hyy2001X, 2021-04-09:
> ok,瞎改以后，19.07的固件终于出来了。
> https://github.com/teasiu/openwrt-autoupdate/releases/tag/AutoUpdate
> 
> 接下来我就作config调整适用编译。
> 有很多补丁和app源码要调整。

已修复并测试通过。当前支持 openwrt/openwrt immortalwrt/immortalwrt coolsnowwolf/lede 这三套源码的常规分支固件生成(18.06 19.07 21.02)

Hyy2001X, 2021-04-09:
[image]

teasiu, 2021-04-09:
great job!

另外，
https://github.com/teasiu/openwrt-autoupdate/blob/master/CustomFiles/Patches/0004-set-default-dnsmasq-full-19.07.patch
https://github.com/teasiu/openwrt-autoupdate/blob/master/CustomFiles/Patches/0004-set-default-dnsmasq-full-21.02.patch
这两个补丁要添加上去。
有一个疑问，
当在原生openwrt19.07编译passwall和ssr时，需要添加lede的基础app到CustomPackages/x86_64，
但是所有脚本都公用一个CustomPackages/x86_64，当需要编译lede的x86_64时，需要去删除内容吧？
参考https://github.com/slitaz-cn/dragino2-19.07.7/blob/main/build.sh里面就有需要添加的基础包。

Thanf-code, 2021-05-18:
> 关于原生openwrt添加ssr和passwall，很多网友编译失败，是因为缺少了upx和ucl.
> 这里将方法写出来，供作者参考利用。
> 手动添加：（原生openwrt源码）
> 1.将lean的lede下tools/upx和tools/ucl拷贝到 原生openwrt源码tools下
> 2.修改tools/Makefile,（33行和46行附近）
> -tools-y += lzma [...]

请教一下如何编译19.07.7固件

teasiu, 2021-05-18:
作者的脚本编译不了19.07.7，只能编译19.07-snpshot
编译19.07.7的方法：
git clone -b v19.07.7 xxx
或者
git clone -b openwrt-19.07 xxx
git chechout v19.07.7 xxx

Hyy2001X, 2021-05-19:
> 作者的脚本编译不了19.07.7，只能编译19.07-snpshot
> 编译19.07.7的方法：
> git clone -b v19.07.7 xxx
> 或者
> git clone -b openwrt-19.07 xxx
> git chechout v19.07.7

已添加 tags 检测,v19.07.7 将识别为19.07

Hyy2001X, 2021-05-19:
> 请教一下如何编译19.07.7固件

修改相关 yml 中的REPO_URL,REPO_BRANCH改为 v19.07

Hyy2001X, 2021-05-19:
> > > 关于原生openwrt添加ssr和passwall，很多网友编译失败，是因为缺少了upx和ucl.
> > > 这里将方法写出来，供作者参考利用。
> > > 手动添加：（原生openwrt源码）
> > > 1.将lean的lede下tools/upx和tools/ucl拷贝到 原生openwrt源码tools下
> > > 2.修改tools/Makefile,（33行和46行附近）
> > > -tools-y += lzma squashfskit4 zip
> > > +tools-y += lzma squashfskit4 zip ucl upx
> > > +$(curdir)/upx/compile := $(curdir)/ucl/compile
> > > 自动添加：（打补丁）
> > > 19.07.7的补丁：
> > > https://raw.githubusercontent.com/slitaz-cn/dragino2-19.07.7/main/patches/0003-add-upx-ucl-for-19.07.7.patch
> > > 21.02的补丁：
> > > https://github.com/fanck0605/openwrt-nanopi-r2s/raw/openwrt-21.02/patches/0004-tools-add-ucl-and-upx-compresss.patch
> > > 缺少上面的补丁，原生openwrt编译ssrp和passwall会失败。
> > > 一定要打上。
> > > 配合scripts/convert_translation.sh
> > > 就可以把大量的18.06的app转过来了。
> > > 希望能帮到大家少走弯路。
> > > 神雕
> > 
> > 
> > 请教一下如何编译19.07.7固件
> 
> 修改相关 yml 中的REPO_URL,REPO_BRANCH改为 v19.07.7

closed 2021-06-27 by Hyy2001X

nnssz, 2021-09-22:
原生openwrt19.07和21.02添加ssr和passwall的脚本补丁
🙏 对于喜欢官方原版openwrt的小白非常有帮助，再次感谢！

nnssz, 2021-09-23:
原生openwrt19.07和21.02添加ssr和passwall的脚本补丁

总结一下，四个补丁要打：
1. upx ucl
2. dnsmasq-full
3. golang
4. 补充一个，wolfssl

Thanf-code, 2021-09-23:
> 原生openwrt19.07和21.02添加ssr和passwall的脚本补丁
> 
> 总结一下，四个补丁要打：
> 
> 1. 添加upx ucl
> 2. dnsmasq更改为dnsmasq-full
> 3. golang版本升级
> 4. 补充一个，将默认的wolfssl改为openssl。
>    [請問大神,如果要用官方源码编译,要改那些地方 #100 (comment)](https://github.com/Hyy2001X/AutoBuild-Actions/issues/100#issuecomment-907805510)

编译成功了吗？

nnssz, 2021-09-23:
我自己还在研究，尝试调整（自己太小白了）。
不过用这位大神https://github.com/starambler/Actions-OpenWrt的actions，可以用openwrt 21.02官方原版源码编译出passwall，供参考。

nnssz, 2021-09-23:
> > 原生openwrt19.07和21.02添加ssr和passwall的脚本补丁
> > 总结一下，四个补丁要打：
> > 
> > 1. 添加upx ucl
> > 2. dnsmasq更改为dnsmasq-full
> > 3. golang版本升级
> > 4. 补充一个，将默认的wolfssl改为openssl。
> >    [請問大神,如果要用官方源码编译,要改那些地方 #100 (comment)](https://github.com/Hyy2001X/AutoBuild-Actions/issues/100#issuecomment-907805510)
> 
> 编译成功了吗？

可以参考这个链接：https://github.com/Hyy2001X/AutoBuild-Actions/issues/100#issuecomment-907805510
此楼主编译成功了。