Hyy2001X/AutoBuild-Actions #64: 几点测试反馈

teasiu, 2021-05-15:
1。CustomPackages/Common第4行
`svn ../feeds/packages/net smartdns https://github.com/kenzok8/openwrt-packages/trunk`
此处应该改为
`svn other smartdns https://github.com/kenzok8/openwrt-packages/trunk`
否则编译无法正确获取该变更。
原因：./feeds/下的文档在install -a的时候，就转如package/feeds/下面了。所以，你install -a后，再修改或变更./feeds/下的文档毫无意义。除非你再 rm -rf ./tmp一次，变更代码后再install -a一次。
当然，因为lede源码里本身在/feeds/packages/net下就有smartdns，才让编译者无感，原生openwrt则直接告诉你缺少smartdns依赖。
2。跟上述有关系，action里ssh调试的时候，.config没有被正确defconfig,后面ssh结束后又加了一次defconfig，
就是说，现在的顺序是
clone --> update&install --> defconfig -->运行scripts -->ssh -->install & defconfig
因为scripts改变了代码，前面的defconfig就没有意义了，那么ssh就只能看看，改了某些配置也无效。
正确顺序是否应该这样：
clone --> update -->运行scripts -->install & defconfig-->ssh
这个顺序的话，前面第一点则维持原状也会生效。
3。openwrt19.07以后，profile已经增加了alias ll和service的支持，建议lede18.06添加上去。
`alias ll='ls -alF --color=auto'
service() {
	[ -f "/etc/init.d/$1" ] || {
		echo "service "'"'"$1"'"'" not found, the following services are available:"
		ls "/etc/init.d"
		return 1
	}
	/etc/init.d/$@
}`
4.对于x86_64固件能否考虑release输出增加vmdk和vhdx
环境增加apt-get install qemu-utils
PS_Firmware的地方
`cd "$GITHUB_WORKSPACE/openwrt/bin"
qemu-img convert -O vhdx ${AutoBuild_Firmware}-Legacy.vmdk ${AutoBuild_Firmware}}-Legacy.vhdx
qemu-img convert -O vhdx ${AutoBuild_Firmware}-UEFI.vmdk ${AutoBuild_Firmware}-UEFI.vhdx`

Hyy2001X, 2021-05-15:
1. 用户应该可以自行编辑这些文件,编译原生 openwrt 的时候删除这行即可，不需要再从 openwrt/package 拉 smartdns 这个包了,当前仅对 lean 源码做了测试
2. 之前不在 run scripts 之前 make defconfig 的话 run scripts 有一些奇奇怪怪的问题,这两天研究一下
3. profile 已修改
4. 由于虚拟磁盘 vmdk和vhdx 这两种不适用于自动更新并且我还没有考虑好以何种形式加入,需要的可以下载软件进行转换,暂时不加入

teasiu, 2021-05-15:
>     1. 用户应该可以自行编辑这些文件,编译原生 openwrt 的时候删除这行即可，不需要再从 openwrt/package 拉 smartdns 这个包了,当前仅对 lean 源码做了测试
> 
>     2. 之前不在 run scripts 之前 make defconfig 的话 run scripts 有一些奇奇怪怪的问题,这两天研究一下
> 
>     3. profile 已修改
> 
>     4. 由于虚拟磁盘 vmdk和vhdx 这两种不适用于自动更新并且我还没有考虑好以何种形式加入,需要的可以下载软件进行转换,暂时不加入

vmdk,vhdx是虚拟磁盘格式，属于导入型，必然不支持内置自动更新，
但是，内置自动更新的更新操作，虚拟系统不受任何影响，（操作无效）。
那么，只需要在自动更新的UI里（或者发布说明里），提示一下“虚拟机导入型磁盘格式时自动更新无效”即可。

发布虚拟机软路由固件，是一个非常节省用户资源的供给侧优化。
我在每一个windows10都安装了hyper-v，都导入一个带翻墙的passwall-openwrt.vhdx
hyper-v设置开机自启512m的openwrt.vhdx
你猜怎么了，嘿嘿，不仅仅是一开机就自然翻墙，而是切切实实的使用openwrt软路由的一切功能。
等于多了一台路由器。

Hyy2001X, 2021-05-19:
> > ```
> > 1. 用户应该可以自行编辑这些文件,编译原生 openwrt 的时候删除这行即可，不需要再从 openwrt/package 拉 smartdns 这个包了,当前仅对 lean 源码做了测试
> > 
> > 2. 之前不在 run scripts 之前 make defconfig 的话 run scripts 有一些奇奇怪怪的问题,这两天研究一下
> > 
> > 3. profile 已修改
> > 
> > 4. 由于虚拟磁盘 vmdk和vhdx 这两种不适用于自动更新并且我还没有考虑好以何种形式加入,需要的可以下载软件进行转换,暂时不加入
> > ```
> 
> vmdk,vhdx是虚拟磁盘格式，属于导入型，必然不支持内置自动更新，
> 但是，内置自动更新的更新操作，虚拟系统不受任何影响，（操作无效）。
> 那么，只需要在自动更新的UI里（或者发布说明里），提示一下“虚拟机导入型磁盘格式时自动更新无效”即可。
> 
> 发布虚拟机软路由固件，是一个非常节省用户资源的供给侧优化。
> 我在每一个windows10都安装了hyper-v，都导入一个带翻墙的passwall-openwrt.vhdx
> hyper-v设置开机自启（仅配置了512m内存）openwrt.vhdx
> 你猜怎么了，嘿嘿，不仅仅是一开机就自然翻墙，而是切切实实的使用openwrt软路由的一切功能。
> 等于多了一台路由器。（当然，hyper的网络设置必须入口外网，出口windows）

已支持上传vhdx vmdk vdi这几种格式的固件

281677160, 2021-06-04:
L大源码，增加个别插件，放package不显示了，要放feeds/luci/applications，依赖放feeds/packages/net，其实很简单的，就是拉取源码后运行一次./scripts/feeds update -a，然后该干嘛就干嘛，就是修改源码啊，增加插件啊啥的，都弄完后运行./scripts/feeds update -a && ./scripts/feeds install -a，接着运行替换.config文件，下面步骤就到SSH修改插件，然后是make defconfig，下来就是判断.config里面的机型了，然后第一次运行定时升级文件把需要的都写入openwrt_info，看了下作者的操作，他是没过SSH就运行文件的，这样如果别人在SSH里面修改了机型就不能正确识别了。

还有就是其实可以把编译主体跟扩展分开的，运行主体的时候下载一下扩展，就跟下载源码意思一样，把扩展下载到指定位置，主体写好到那里运行什么就可以，这样别人拉取你的仓库每次编译就是最新修改的扩展文件了，不需要频繁的合拼仓库或者重新拉取仓库的，太麻烦了

可以仓库下我的仓库 https://github.com/281677160/AutoBuild-OpenWrt

我正在弄本地一键编译，把定时升级固件都弄进去，不过编译10次，有9次都出错的，要么直接源码都下载错误，要么下载DL错误，不要这些东西，就源码本身，再加一个插件包的话，那就10次编译能成功9次了，搞的我头大了，不过其实弄太多这些也没必要，本地编译要定时升级还得把固件又传到github，烦

closed 2021-07-11 by Hyy2001X