Hyy2001X/AutoBuild-Actions #67: 请教一下 编译newifi3v1.2（闪存32M 内存512M）19.07.7要改哪些地方？

Thanf-code, 2021-05-18:
编译 新路由3v1.2（闪存32M 内存512M）的openwrt19.07.7版  要改动哪些地方？

Hyy2001X, 2021-05-19:
修改 yml 文件,REPO_URL 改为 https://github.com/openwrt/openwrt/
REPO_BRANCH 改为 v19.07.7
修改 Scripts/AutoBuild_DiyScript.sh 中的 INCLUDE_Obsolete_PKG_Compatible 为 true

Thanf-code, 2021-05-20:
[image]

> 先同步我的最新改动,然后
> 修改 yml 文件,REPO_URL 改为 https://github.com/openwrt/openwrt/
> REPO_BRANCH 改为 v19.07.7
> 修改 Scripts/AutoBuild_DiyScript.sh 中的 INCLUDE_Obsolete_PKG_Compatible 为 true

编译时出现如下错误， 编译完成后有影响吗？
[image]

Thanf-code, 2021-05-20:
> 先同步我的最新改动,然后
> 修改 yml 文件,REPO_URL 改为 https://github.com/openwrt/openwrt/
> REPO_BRANCH 改为 v19.07.7
> 修改 Scripts/AutoBuild_DiyScript.sh 中的 INCLUDE_Obsolete_PKG_Compatible 为 true

编译是出现如下错误，编译完成后有影响吗？
[image]

Hyy2001X, 2021-05-20:
> > 先同步我的最新改动,然后
> > 修改 yml 文件,REPO_URL 改为 https://github.com/openwrt/openwrt/
> > REPO_BRANCH 改为 v19.07.7
> > 修改 Scripts/AutoBuild_DiyScript.sh 中的 INCLUDE_Obsolete_PKG_Compatible 为 true
> 
> 编译是出现如下错误，编译完成后有影响吗？
> [image]

没有影响,无视即可

Thanf-code, 2021-05-20:
> > > 先同步我的最新改动,然后
> > > 修改 yml 文件,REPO_URL 改为 https://github.com/openwrt/openwrt/
> > > REPO_BRANCH 改为 v19.07.7
> > > 修改 Scripts/AutoBuild_DiyScript.sh 中的 INCLUDE_Obsolete_PKG_Compatible 为 true
> > 
> > 
> > 编译是出现如下错误，编译完成后有影响吗？
> > [image]
> 
> 没有影响,无视即可

无法ssh登录
[image]

Thanf-code, 2021-05-21:
> > > > 先同步我的最新改动,然后
> > > > 修改 yml 文件,REPO_URL 改为 https://github.com/openwrt/openwrt/
> > > > REPO_BRANCH 改为 v19.07.7
> > > > 修改 Scripts/AutoBuild_DiyScript.sh 中的 INCLUDE_Obsolete_PKG_Compatible 为 true
> > > 
> > > 
> > > 编译是出现如下错误，编译完成后有影响吗？
> > > [image]
> > 
> > 
> > 没有影响,无视即可
> 
> 无法ssh登录
> [image]

可以登录 就是耗时有点长

closed 2021-05-26 by Hyy2001X