Hyy2001X/AutoBuild-Actions #73: 请问0506版如何手动升级至0608版？

iJayscn, 2021-06-12:
0506一键更新找不到新版固件。

0608固件已下载。

iJayscn, 2021-06-12:
使用sysupgrade -v 命令貌似升级成功了，不知有啥后遗症没有。目前是进入固件ssh欢迎界面，没有固件版本信息，如下：
root@OpenWrt:/tmp# sysupgrade -v

          ——————————————————————————————————————————————————
             Openwrt R21.6.1-20210608 / AutoUpdate V6.0.4
          ——————————————————————————————————————————————————

           设备名称:		d-team_newifi-d2
           固件版本:		
           IP 地址: 		192.168.1.1
           运行时间: 		0 天 0 小时 3 分钟
           CoreMark 得分: 	0
           可用空间:		4.4M

Hyy2001X, 2021-06-12:
没有问题，如果有强迫症的话建议执行一下 sed -i "/profile/d" /lib/upgrade/keep.d/base-files-essential,然后再更新一遍固件

iJayscn, 2021-06-12:
> 没有问题，如果有强迫症的话建议执行一下 sed -i "/profile/d" /lib/upgrade/keep.d/base-files-essential,然后再更新一遍固件

谢谢

closed 2021-06-12 by iJayscn