Hyy2001X/AutoBuild-Actions #79: 使用你的脚本,

xm3good, 2021-06-22:
使用你的脚本,我的是华硕ACRH17路由器,怎么在你的脚本里添加设备.

Hyy2001X, 2021-06-22:
直接按照教程操作编译即可,不用在脚本里添加设备

xm3good, 2021-06-24:
快整哭了,还是不行
20s
Run chmod +x Scripts/AutoBuild_*.sh
  chmod +x Scripts/AutoBuild_*.sh
  cp $GITHUB_WORKSPACE/Configs/$DEFAULT_TARGET openwrt/.config
  cd openwrt
  make defconfig > /dev/null 2>&1
  source $GITHUB_WORKSPACE/Scripts/AutoBuild_DiyScript.sh
  source $GITHUB_WORKSPACE/Scripts/AutoBuild_Function.sh
  Firmware-Diy_Base
  source $GITHUB_WORKSPACE/openwrt/VARIABLE_FILE_Sec && Firmware-Diy
  rm -f .config && cp $GITHUB_WORKSPACE/Configs/$DEFAULT_TARGET .config
  shell: /usr/bin/bash -e {0}
  env:
    DEFAULT_TARGET: ipq40xx
    DEFAULT_SOURCE: coolsnowwolf/lede:master
    UPLOAD_RELEASES: true
    UPLOAD_ARTIFACTS: false
    UPLOAD_BIN_ARTIFACTS: false
    DELETE_USELESS_FILES: true
    DELETE_OLD_WORKFLOW: true
    REPO_URL: https://github.com/coolsnowwolf/lede
    REPO_BRANCH: master
    Artifacts_Date: 06241417
Error: 28] [ERROR] Can not obtain the TARGET_PROFILE !
Error: Process completed with exit code 1.
也在config目录添加ipq40xx文件了啊

Hyy2001X, 2021-06-24:
收到，我测试一下

Hyy2001X, 2021-06-24:
参考一下
https://github.com/Hyy2001X/AutoBuild-Actions-BETA/commit/0a00cd90378906c685fba4c330858f0e1db8b6c0
https://github.com/Hyy2001X/AutoBuild-Actions-BETA/commit/1b118abe1694fcf6703c092c0547db3652da1ad0

Hyy2001X, 2021-06-24:
已添加 asus_rt-acrh17 支持,fork 之后按需修改即可

closed 2021-06-24 by Hyy2001X