Hyy2001X/AutoBuild-Actions #80: 同步最新源码编译 lan口无法连接

Paris0-0, 2021-06-22:
18号的时候同步了次你的源码编译出来后恢复配置功能正常但怎么都登不上管理界面,密码是对的但一直报错,ssh可以登陆
今天又同步了次源码,删除配置全新刷机,lan口无法连接,无论dhcp分配还是手动指定ip都无法连接路由,只能通过WiFi连接,wan口也不能拨号.
回滚13号编译的固件恢复配置后正常.
而且之前几次每次升级都会遇到登陆界面argon主题布局异常,没有背景图,然后第一次关机重启后配置全部丢失

Hyy2001X, 2021-06-22:
可能是修改过 IP 地址的问题,试试不修改 IP 编译再刷入

Hyy2001X, 2021-06-22:
先同步一下我的改动,我禁用了一些 IP修改功能

Paris0-0, 2021-06-22:
IP我还是需要修改的，等这部分功能稳定我再试试，天太热了蹲着刷固件快热死了

281677160, 2021-06-23:
> 可能是修改过 IP 地址的问题,试试不修改 IP 编译再刷入

修改IP最好是把代码写入zzz-default-settings文件，写2行代码

uci set network.lan.ipaddr='192.168.2.2'
uci commit network

就这样的两行，改IP行在上面，这样修改IP比较稳妥，特别是Lienol大的源码，不这样修改IP会有问题的

Hyy2001X, 2021-06-23:
> > 可能是修改过 IP 地址的问题,试试不修改 IP 编译再刷入
> 
> 修改IP最好是把代码写入zzz-default-settings文件，写2行代码
> 
> uci set network.lan.ipaddr='192.168.2.2'
> uci commit network
> 
> 就这样的两行，改IP行在上面，这样修改IP比较稳妥，特别是Lienol大的源码，不这样修改IP会有问题的

好的感谢

Paris0-0, 2021-06-23:
我以前手动编译一直是改这个文件，那看来sed改一下就行了

281677160, 2021-06-23:
> 我以前手动编译一直是改这个文件，那看来sed改一下就行了

我的改一堆，X86旁路由，我编译出来的固件都不需要设置的，所以作者的定时更新固件应该要做一个可以选择不保留配置更新按钮才好，哈哈

sed -i "/uci commit fstab/a\uci commit network" $ZZZ
sed -i "/uci commit network/i\uci set network.lan.ipaddr='192.168.2.2'" $ZZZ                      # IPv4 地址(openwrt后台地址)
sed -i "/uci commit network/i\uci set network.lan.netmask='255.255.255.0'" $ZZZ                   # IPv4 子网掩码
sed -i "/uci commit network/i\uci set network.lan.gateway='192.168.2.1'" $ZZZ                     # IPv4 网关
sed -i "/uci commit network/i\uci set network.lan.broadcast='192.168.2.255'" $ZZZ                 # IPv4 广播
sed -i "/uci commit network/i\uci set network.lan.dns='223.5.5.5 114.114.114.114'" $ZZZ           # DNS(多个DNS要用空格分开)
sed -i "/uci commit network/i\uci set network.lan.delegate='0'" $ZZZ                              # 去掉LAN口使用内置的 IPv6 管理

$ZZZ 是路径来的

Paris0-0, 2021-06-23:
路径可以用作者脚本里的PKG_Finder,不过我看他昨天更新Readme把这个去掉了😂

Hyy2001X, 2021-06-23:
> 路径可以用作者脚本里的PKG_Finder,不过我看他昨天更新Readme把这个去掉了😂

感觉写的太烂 去掉了

Paris0-0, 2021-06-27:
同步了最新的源码，还是有问题，安装完第一次登陆界面布局出错没有背景,提示没有设置密码,用老密码能登进去且功能正常,但第一次重启后配置全部丢失,只能手动恢复备份...还是改ip的原因吗?

Hyy2001X, 2021-06-27:
> 同步了最新的源码，还是有问题，安装完第一次登陆界面布局出错没有背景,提示没有设置密码,用老密码能登进去且功能正常,但第一次重启后配置全部丢失,只能手动恢复备份...还是改ip的原因吗?

手头上没有新三,x64不改ip一切正常

Paris0-0, 2021-06-27:
刚又试了下不改IP,Default_LAN_IP=false,问题一模一样,第一次提示未设置密码实际可登陆可用,重启丢失所有配置
[image]

Hyy2001X, 2021-06-27:
> 刚又试了下不改IP,Default_LAN_IP=false,问题一模一样,刚刷完提示未设置密码实际可登陆,配置都在可上网,第一次重启丢失所有配置,恢复配置后续重启正常不会丢配置
> [image]

luci界面很奇怪,版本号也没有正确显示,需要更多的信息

Paris0-0, 2021-06-27:
这就是每次升级完后第一次进入的登陆界面,只有这里是异常的,其他都完全正常配置都在;
手动重启后完全丢失配置,这个登录界面也恢复初始;
恢复配置后也无异常再重启也不会丢失配置
我现在已经恢复过配置了一切正常.记不得从什么时候开始每次升级都是这样的界面,重启丢配置了。。
我前两天已经纯净刷机然后把所有配置都在luci界面配置一遍，也不存在旧的配置冲突什么了
[image]

Paris0-0, 2021-06-27:
就这些吧,编译出来28-29MB
[image]

Paris0-0, 2021-07-11:
换用你仓库里精简的config后正常了，之前用的是源码生成的config，可能比较老有冲突。
另问下现在手动运行workflow时候填的lan ip生效吗？之前是不是取消了

closed 2021-07-11 by Paris0-0