Hyy2001X/AutoBuild-Actions #83: 新路由3 v1.2编译官方openwrt失败

Thanf-code, 2021-06-26:
[logs_9.zip](https://github.com/Hyy2001X/AutoBuild-Actions/files/6719844/logs_9.zip)

Hyy2001X, 2021-06-26:
https://github.com/Thanf-code/AutoBuild-Actions/blob/master/CustomPackages/d-team_newifi-d2
注释掉 mt76 那行试试,要是还不行的话我也不知道了

Thanf-code, 2021-06-27:
> https://github.com/Thanf-code/AutoBuild-Actions/blob/master/CustomPackages/d-team_newifi-d2
> 注释掉 mt76 那行试试,要是还不行的话我也不知道了

[image]
[image]

281677160, 2021-06-27:
你这个缺很多依赖啊，你你现在编译进固件的smartdns就缺依赖出错
[image]

Hyy2001X, 2021-06-27:
> 你这个缺很多依赖啊，你你现在编译进固件的smartdns就缺依赖出错
> [image]

缺依赖问题自己解决

wmyfelix, 2021-06-27:
> https://github.com/Thanf-code/AutoBuild-Actions/blob/master/CustomPackages/d-team_newifi-d2
> 注释掉 mt76 那行试试,要是还不行的话我也不知道了
最近固件发现无线无法驱动了，但是我config里带着mt76编译的，是驱动失效了么，我可以通过把闭源驱动取消注释直接使用么？

Hyy2001X, 2021-06-27:
> > https://github.com/Thanf-code/AutoBuild-Actions/blob/master/CustomPackages/d-team_newifi-d2
> > 注释掉 mt76 那行试试,要是还不行的话我也不知道了
> 
> 最近固件发现无线无法驱动了，但是我config里带着mt76编译的，是驱动失效了么，我可以通过把闭源驱动取消注释直接使用么？
> 保留配置升级没有问题，最近在breed里升级后无线消失

无线消失是指LUCI上的系统-无线这个选项消失吗

wmyfelix, 2021-06-27:
> > > https://github.com/Thanf-code/AutoBuild-Actions/blob/master/CustomPackages/d-team_newifi-d2
> > > 注释掉 mt76 那行试试,要是还不行的话我也不知道了
> > 
> > 
> > 最近固件发现无线无法驱动了，但是我config里带着mt76编译的，是驱动失效了么，我可以通过把闭源驱动取消注释直接使用么？
> > 保留配置升级没有问题，最近在breed里升级后无线消失
> 
> 无线消失是指LUCI上的系统-无线这个选项消失吗

是的

Hyy2001X, 2021-06-27:
> > > > https://github.com/Thanf-code/AutoBuild-Actions/blob/master/CustomPackages/d-team_newifi-d2
> > > > 注释掉 mt76 那行试试,要是还不行的话我也不知道了
> > > 
> > > 
> > > 最近固件发现无线无法驱动了，但是我config里带着mt76编译的，是驱动失效了么，我可以通过把闭源驱动取消注释直接使用么？
> > > 保留配置升级没有问题，最近在breed里升级后无线消失
> > 
> > 
> > 无线消失是指LUCI上的系统-无线这个选项消失吗
> 
> 是的

看了一下你项目里的文件都比较老了例如profile mac80211.sh,都需要适配最新的源码,或者同步我仓库里的那几个文件

281677160, 2021-06-27:
> 看了一下你项目里的文件都比较老了例如profile mac80211.sh,都需要适配最新的源码,或者同步我仓库里的那几个文件

大神，我都给过你建议了，主体，扩展文件，插件包，应该分开搞的，别人拉取你的仓库就拉取主体就可以了，插件包跟扩展文件每次别人启动主体（.yml）的时候就下载进去，这样就不会造成拉你仓库的人老是要去同步你仓库，主体不修改代码的情况下，别人根本不需要去考虑同步你仓库的问题，因为每次编译都是你最新的扩展文件跟插件

wmyfelix, 2021-06-27:
> 看了一下你项目里的文件都比较老了例如profile mac80211.sh,都需要适配最新的源码,或者同步我仓库里的那几个文件

谢谢答复。 `mac80211.sh` ,这个的上游具体在哪里啊，为什么要替换这个文件

Hyy2001X, 2021-06-27:
> 谢谢答复。 `mac80211.sh` ,这个的上游具体在哪里啊，为什么要替换这个文件

就在是 lede 源码里面的,我只是单独拿出来并做了一些修改,wifi名称啥的

Hyy2001X, 2021-06-27:
> 大神，我都给过你建议了，主体，扩展文件，插件包，应该分开搞的，别人拉取你的仓库就拉取主体就可以了，插件包跟扩展文件每次别人启动主体（.yml）的时候就下载进去，这样就不会造成拉你仓库的人老是要去同步你仓库，主体不修改代码的情况下，别人根本不需要去考虑同步你仓库的问题，因为每次编译都是你最新的扩展文件跟插件

每个人修改的内容和需求都不一样的,拉取的内容就不能100%确定,还是有待考虑

wmyfelix, 2021-06-27:
> 大神，我都给过你建议了，主体，扩展文件，插件包，应该分开搞的，别人拉取你的仓库就拉取主体就可以了，插件包跟扩展文件每次别人启动主体（.yml）的时候就下载进去，这样就不会造成拉你仓库的人老是要去同步你仓库，主体不修改代码的情况下，别人根本不需要去考虑同步你仓库的问题，因为每次编译都是你最新的扩展文件跟插件

这是我自己的repo，因为之前fork的upstream删了哈哈

wmyfelix, 2021-06-27:
> 就在是 lede 源码里面的,我只是单独拿出来并做了一些修改,wifi名称啥的

这样的话感觉用sed之类的patch会更合适

281677160, 2021-06-27:
> 每个人修改的内容和需求都不一样的,拉取的内容就不能100%确定,还是有待考虑

呃，他们可以有他们的需求啊，有扩展文件给他们修改他们自己需要的，你的扩展只是把源码修复好，整理好而已，就拿P3大神的脚本来说，他的就是最原始的主体，啥都没的，就编译，但是源码多了，机型多了，很多都需要你来适配的，你就是做的适配工作或者源码那里需要修改过更好的，都是你来做，他们不是没办法修改东西啊，还有你的定时更新脚本，其他的东西，不是都是你做的吗？这些东西每次编译的时候都是你最新修改的文件，不舒服吗？所有开关那些，你可以在主体仓库做一个文件专门设置开关的，运行.yml的时候读取一下就好了，这样他们不是可以自己喜欢开啥关啥的么？他们有DIY.SH的扩展文件，想弄啥就弄啥

wmyfelix, 2021-06-27:
> 呃，他们可以有他们的需求啊，有扩展文件给他们修改他们自己需要的，你的扩展只是把源码修复好，整理好而已，就拿P3大神的脚本来说，他的就是最原始的主体，啥都没的，就编译，但是源码多了，机型多了，很多都需要你来适配的，你就是做的适配工作或者源码那里需要修改过更好的，都是你来做，他们不是没办法修改东西啊，还有你的定时更新脚本，其他的东西，不是都是你做的吗？这些东西每次编译的时候都是你最新修改的文件，不舒服 [...]

这些开源的东西还是作者怎么开心怎么来吧，没必要把自己想法强加个别人。实在不行你可以自己弄然后提pr ((
还想问一下config里的软件包是可以自己解决一定的依赖关系吗，感觉即便功能相同每个人的include=y的数量也不一样。
例如samba这里好像没有samba-server只有autosamba和luci

281677160, 2021-06-27:
> 这些开源的东西还是作者怎么开心怎么来吧，没必要把自己想法强加个别人。实在不行你可以自己弄然后提pr ((
> 还想问一下config里的软件包是可以自己解决一定的依赖关系吗，感觉即便功能相同每个人的include=y的数量也不一样。
> 例如samba这里好像没有samba-server只有autosamba和luci

呃，我早差不多半年前就这样做了，只是给大神一个建议而已，他的现在这样太麻烦，人家用他仓库老是要同步，要不然他的扩展你们跟不上， config不能解决任何依赖问题，解决依赖是靠插件包，或者说你的插件少什么依赖就要去拉取什么依赖，你说的samba的samba-server文件在network选项里面

281677160, 2021-06-27:
我的仓库就是主体，扩展，插件包分开组成的，不过我现在也不怎么搞而已，我以前专门教小白编译的，任何用过我仓库的人都说方便，易用，简单，不过我没技术，好多东西知道有问题就是不知道怎么写代码解决而已，哈哈，其实这些都不是事，反正都小白用，大家用的方便就好了，不过我现在不搞了，太累，实在是没技术，我删仓库都老久了，现在的仓库是我新建的

wmyfelix, 2021-06-27:
> 呃，我早差不多半年前就这样做了，只是给大神一个建议而已，他的现在这样太麻烦，人家用他仓库老是要同步，要不然他的扩展你们跟不上， config不能解决任何依赖问题，解决依赖是靠插件包，或者说你的插件少什么依赖就要去拉取什么依赖，你说的samba的samba-server文件在network选项里面

config文件不是make menuconfig生成的么，如果不能处理包与包间的依赖关系，samba-server也是要在那里选上的吧。但是这个[config](https://github.com/Hyy2001X/AutoBuild-Actions/blob/master/Configs/d-team_newifi-d2)里没有include samba-server啊，只有
`CONFIG_PACKAGE_autosamba=y
CONFIG_PACKAGE_luci-app-samba=y`

281677160, 2021-06-27:
> config文件不是make menuconfig生成的么，如果不能处理包与包间的依赖关系，samba-server也是要在那里选上的吧。但是这个[config](https://github.com/Hyy2001X/AutoBuild-Actions/blob/master/Configs/d-team_newifi-d2)里没有include samba-server啊，只有 [...]

一般情况下，你在make menuconfig选择插件的时候是不需要考虑这些的，你只要把所有luci-app-xxx选好就行了，你说的samba-server是你选择了luci-app-samba就自动勾选的，这个你自己要看的话，就在network选项里面啊，刚才不是说了么？如果你的插件缺依赖，一般情况都不在这里处理，而是你的插件缺少文件，缺少文件这里处理不了的

281677160, 2021-06-27:
> config文件不是make menuconfig生成的么，如果不能处理包与包间的依赖关系，samba-server也是要在那里选上的吧。但是这个[config](https://github.com/Hyy2001X/AutoBuild-Actions/blob/master/Configs/d-team_newifi-d2)里没有include samba-server啊，只有 [...]

你看上面的图片了吗？我另外截了小图，那些就是缺依赖显示，就是缺少了文件了，如果你编译的时候选择了该缺依赖的插件，你编译100%出错

wmyfelix, 2021-06-27:
> 你看上面的图片了吗？我另外截了小图，那些就是缺依赖显示，就是缺少了文件了，如果你编译的时候选择了该缺依赖的插件，你编译100%出错

感谢耐心答复。  
我没有亲自make menuconfig过，我说的config指的就是超链接中的那个，那里面没有类似`CONFIG_PACKAGE_samba36-server=y`的字段，但成功编译并在生成固件中有samba36-server
我没有在意samba-server是在什么选项中，而是想知道如果在make menuconfig过程中选中了为什么`CONFIG_PACKAGE_samba36-server=y`没有出现在其所生成的config文件里。  
难道network选项是别的什么东西...

Hyy2001X, 2021-06-27:
> 感谢耐心答复。
> 我没有亲自make menuconfig过，我说的config指的就是超链接中的那个，那里面没有类似`CONFIG_PACKAGE_samba36-server=y`的字段，但成功编译并在生成固件中有samba36-server
> 我没有在意samba-server是在什么选项中，而是想知道如果在make [...]

仓库里面的那些.config文件是精简过的,为了方便修改。当你在本地make defconfig 或者 make menuconfig 以后所有的内容就都会补全了

Hyy2001X, 2021-06-27:
> > > > > > > > > > > > > > https://github.com/Thanf-code/AutoBuild-Actions/blob/master/CustomPackages/d-team_newifi-d2
> > > > > > > > > > > > > > 注释掉 mt76 那行试试,要是还不行的话我也不知道了
> > > > > > > > > > > > > 
> > > > > > > > > > > > > 
> > > > > > > > > > > > > 最近固件发现无线无法驱动了，但是我config里带着mt76编译的，是驱动失效了么，我可以通过把闭源驱动取消注释直接使用么？
> > > > > > > > > > > > > 保留配置升级没有问题，最近在breed里升级后无线消失
> > > > > > > > > > > > 
> > > > > > > > > > > > 
> > > > > > > > > > > > 无线消失是指LUCI上的系统-无线这个选项消失吗
> > > > > > > > > > > 
> > > > > > > > > > > 
> > > > > > > > > > > 是的
> > > > > > > > > > 
> > > > > > > > > > 
> > > > > > > > > > 看了一下你项目里的文件都比较老了例如profile mac80211.sh,都需要适配最新的源码,或者同步我仓库里的那几个文件
> > > > > > > > > 
> > > > > > > > > 
> > > > > > > > > 大神，我都给过你建议了，主体，扩展文件，插件包，应该分开搞的，别人拉取你的仓库就拉取主体就可以了，插件包跟扩展文件每次别人启动主体（.yml）的时候就下载进去，这样就不会造成拉你仓库的人老是要去同步你仓库，主体不修改代码的情况下，别人根本不需要去考虑同步你仓库的问题，因为每次编译都是你最新的扩展文件跟插件
> > > > > > > > 
> > > > > > > > 
> > > > > > > > 每个人修改的内容和需求都不一样的,拉取的内容就不能100%确定,还是有待考虑
> > > > > > > 
> > > > > > > 
> > > > > > > 呃，他们可以有他们的需求啊，有扩展文件给他们修改他们自己需要的，你的扩展只是把源码修复好，整理好而已，就拿P3大神的脚本来说，他的就是最原始的主体，啥都没的，就编译，但是源码多了，机型多了，很多都需要你来适配的，你就是做的适配工作或者源码那里需要修改过更好的，都是你来做，他们不是没办法修改东西啊，还有你的定时更新脚本，其他的东西，不是都是你做的吗？这些东西每次编译的时候都是你最新修改的文件，不舒服吗？所有开关那些，你可以在主体仓库做一个文件专门设置开关的，运行.yml的时候读取一下就好了，这样他们不是可以自己喜欢开啥关啥的么？他们有DIY.SH的扩展文件，想弄啥就弄啥
> > > > > > 
> > > > > > 
> > > > > > 这些开源的东西还是作者怎么开心怎么来吧，没必要把自己想法强加个别人。实在不行你可以自己弄然后提pr ((
> > > > > > 还想问一下config里的软件包是可以自己解决一定的依赖关系吗，感觉即便功能相同每个人的include=y的数量也不一样。
> > > > > > 例如samba这里好像没有samba-server只有autosamba和luci
> > > > > 
> > > > > 
> > > > > 呃，我早差不多半年前就这样做了，只是给大神一个建议而已，他的现在这样太麻烦，人家用他仓库老是要同步，要不然他的扩展你们跟不上， config不能解决任何依赖问题，解决依赖是靠插件包，或者说你的插件少什么依赖就要去拉取什么依赖，你说的samba的samba-server文件在network选项里面
> > > > 
> > > > 
> > > > config文件不是make menuconfig生成的么，如果不能处理包与包间的依赖关系，samba-server也是要在那里选上的吧。但是这个[config](https://github.com/Hyy2001X/AutoBuild-Actions/blob/master/Configs/d-team_newifi-d2)里没有include samba-server啊，只有
> > > > `CONFIG_PACKAGE_autosamba=y CONFIG_PACKAGE_luci-app-samba=y`
> > > 
> > > 
> > > 你看上面的图片了吗？我另外截了小图，那些就是缺依赖显示，就是缺少了文件了，如果你编译的时候选择了该缺依赖的插件，你编译100%出错
> > 
> > 
> > 感谢耐心答复。
> > 我没有亲自make menuconfig过，我说的config指的就是超链接中的那个，那里面没有类似`CONFIG_PACKAGE_samba36-server=y`的字段，但成功编译并在生成固件中有samba36-server
> > 我没有在意samba-server是在什么选项中，而是想知道如果在make menuconfig过程中选中了为什么`CONFIG_PACKAGE_samba36-server=y`没有出现在其所生成的config文件里。
> > 难道network选项是别的什么东西...
> 
> 仓库里面的那些.config文件是精简过的,为了方便修改。当你在本地make defconfig 或者 make menuconfig 以后所有的内容就都会补全了，之所以没看到CONFIG_PACKAGE_samba36-server=y,因为这个包是默认被选中的,不需要额外在config中额外添加

wmyfelix, 2021-06-27:
> 仓库里面的那些.config文件是精简过的,为了方便修改。当你在本地make defconfig 或者 make menuconfig 以后所有的内容就都会补全了

对啊这个我知道，但是用精简过的为什么也能正常编译，所以我才猜测能解决一些依赖关系

281677160, 2021-06-27:
> 对啊这个我知道，但是用精简过的为什么也能正常编译，所以我才猜测能解决一些依赖关系

这个只是仓库的一个文件而已，真正的文件在源码里面，这个仓库的.config的内容是写入源码里面的.config里去，只要luci-app-xxx存在，他就自动补齐本身这个插件所需的依赖全部东西，只要你源码里面有，如果没有就是缺依赖，仓库的这个.config的作用就等于make menuconfig的时候选插件啊，你选插件的时候不是只看到luci-app-xxx吗？你一次都没进过make menuconfig的话，跟你说这个你根本听不懂，如果你的源码缺依赖，你能在.config里面解决什么依赖?

wmyfelix, 2021-06-27:
我说的“解决依赖”就是这个
> 他就自动补齐本身这个插件所需的依赖全部东西
我确实对Make的过程一窍不通🙌
感谢大佬指点

> 这个只是仓库的一个文件而已，真正的文件在源码里面，这个仓库的.config的内容是写入源码里面的.config里去，只要luci-app-xxx存在，他就自动补齐本身这个插件所需的依赖全部东西，只要你源码里面有，如果没有就是缺依赖，仓库的这个.config的作用就等于make menuconfig的时候选插件啊，你选插件的时候不是只看到luci-app-xxx吗？你一次都没进过make [...]

281677160, 2021-06-27:
> 我说的“解决依赖”就是这个
>
>
> 我确实对Make的过程一窍不通🙌
> 感谢大佬指点

仓库的.config这个文件就=make menuconfig选择插件，一个是自己写的，一个是图形界面选择的，它的最终目的都是把内容写入源码的.config文件里面，源码的.config有你需要的插件就自动拉取依赖了，仓库的.config可以随便叫什么名字都可以，只要你写的内容是对的就可以了，有好多人的仓库就不需要SSH连接进行make menuconfig选择插件，就靠写仓库的.config，我以前教人家云编译的时候也试过，不过这个对小白来说更高难度，要知道每个插件的正确名字才可以写出正确的.config文件，有图形界面的干嘛不用图形界面的呢？

Thanf-code, 2021-06-27:
> > https://github.com/Thanf-code/AutoBuild-Actions/blob/master/CustomPackages/d-team_newifi-d2
> > 注释掉 mt76 那行试试,要是还不行的话我也不知道了
> 
> [image]
> [image]
> 
> 再注释掉smartdns相关内容

谢谢大神回复  注释掉smartdns后  编译成功了，能教教我怎么解决依赖问题吗？

281677160, 2021-06-27:
> > > https://github.com/Thanf-code/AutoBuild-Actions/blob/master/CustomPackages/d-team_newifi-d2
> > > 注释掉 mt76 那行试试,要是还不行的话我也不知道了
> > 
> > 
> > [image]
> > [image]
> > 再注释掉smartdns相关内容
> 
> 谢谢大神回复 注释掉smartdns后 编译成功了，能教教我怎么解决依赖问题吗？

你这个缺的不是一个依赖啊，是好几个，其他的你要根据你拉的什么插件去找，samrtdns的依赖好解决，前提是要知道你什么源码的，因为这个插件有适配18.06跟19.07或者以上的，就是luci不一样，依赖应该是一样的

281677160, 2021-06-27:
> 你这个缺的不是一个依赖啊，是好几个，其他的你要根据你拉的什么插件去找，samrtdns的依赖好解决，前提是要知道你什么源码的，因为这个插件有适配18.06跟19.07或者以上的，就是luci不一样，依赖应该是一样的

18.06的或者以下的用这个
git clone -b lede https://github.com/pymumu/luci-app-smartdns package/luci-app-smartdns
git clone https://github.com/pymumu/openwrt-smartdns package/smartdns

19.07的或者以上的用这个
git clone https://github.com/pymumu/luci-app-smartdns package/luci-app-smartdns
git clone https://github.com/pymumu/openwrt-smartdns package/smartdns

closed 2021-06-27 by Hyy2001X