Hyy2001X/AutoBuild-Actions #84: 华硕AC58U编失败

xm3good, 2021-07-01:
不知道为什么这两天编AC58会失败了,正常来说 AC58和ACRH17两个都一样,后者就能过去,前者前两天还能正常编过,
不知道怎么传编译文件,就复制一点吧
mkdir -p /home/runner/work/AutoBuild-Actions-Template/AutoBuild-Actions-Template/openwrt/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-ipq40xx/stamp
SHELL= flock /home/runner/work/AutoBuild-Actions-Template/AutoBuild-Actions-Template/openwrt/tmp/.root-copy.flock -c 'cp -fpR /home/runner/work/AutoBuild-Actions-Template/AutoBuild-Actions-Template/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/ppp-default/ppp-2.4.8.git-2020-09-04/.pkgdir/ppp/. /home/runner/work/AutoBuild-Actions-Template/AutoBuild-Actions-Template/openwrt/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-ipq40xx/'
touch /home/runner/work/AutoBuild-Actions-Template/AutoBuild-Actions-Template/openwrt/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-ipq40xx/stamp/.ppp_installed
echo "ppp" >> /home/runner/work/AutoBuild-Actions-Template/AutoBuild-Actions-Template/openwrt/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/pkginfo/ppp.default.install
mkdir -p /home/runner/work/AutoBuild-Actions-Template/AutoBuild-Actions-Template/openwrt/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-ipq40xx/stamp
SHELL= flock /home/runner/work/AutoBuild-Actions-Template/AutoBuild-Actions-Template/openwrt/tmp/.root-copy.flock -c 'cp -fpR /home/runner/work/AutoBuild-Actions-Template/AutoBuild-Actions-Template/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/ppp-default/ppp-2.4.8.git-2020-09-04/.pkgdir/ppp-mod-pppoe/. /home/runner/work/AutoBuild-Actions-Template/AutoBuild-Actions-Template/openwrt/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-ipq40xx/'
touch /home/runner/work/AutoBuild-Actions-Template/AutoBuild-Actions-Template/openwrt/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-ipq40xx/stamp/.ppp-mod-pppoe_installed
echo "ppp-mod-pppoe" >> /home/runner/work/AutoBuild-Actions-Template/AutoBuild-Actions-Template/openwrt/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/pkginfo/ppp.default.install
make[3]: Leaving directory '/home/runner/work/AutoBuild-Actions-Template/AutoBuild-Actions-Template/openwrt/package/network/services/ppp'
time: package/network/services/ppp/default/compile#0.21#0.06#0.25
make[3]: Entering directory '/home/runner/work/AutoBuild-Actions-Template/AutoBuild-Actions-Template/openwrt/package/other/helloworld/shadowsocksr-libev'
mkdir -p /home/runner/work/AutoBuild-Actions-Template/AutoBuild-Actions-Template/openwrt/dl
SHELL= flock /home/runner/work/AutoBuild-Actions-Template/AutoBuild-Actions-Template/openwrt/tmp/.shadowsocksr-libev-2.5.6.tar.xz.flock -c '  	/home/runner/work/AutoBuild-Actions-Template/AutoBuild-Actions-Template/openwrt/scripts/download.pl "/home/runner/work/AutoBuild-Actions-Template/AutoBuild-Actions-Template/openwrt/dl" "shadowsocksr-libev-2.5.6.tar.xz" "34308ed827a5dd4f4e35619914102d55b00604faa44fda051d1d25fb4a319325" "" || (  /home/runner/work/AutoBuild-Actions-Template/AutoBuild-Actions-Template/openwrt/scripts/dl_github_archive.py --dl-dir="/home/runner/work/AutoBuild-Actions-Template/AutoBuild-Actions-Template/openwrt/dl" --url="https://github.com/shadowsocksrr/shadowsocksr-libev" --version="d63ff863800a5645aca4309d5dd5962bd1e95543" --subdir="shadowsocksr-libev-2.5.6" --source="shadowsocksr-libev-2.5.6.tar.xz" --hash="34308ed827a5dd4f4e35619914102d55b00604faa44fda051d1d25fb4a319325" || ( 	echo "Checking out files from the git repository..."; mkdir -p /home/runner/work/AutoBuild-Actions-Template/AutoBuild-Actions-Template/openwrt/tmp/dl && cd /home/runner/work/AutoBuild-Actions-Template/AutoBuild-Actions-Template/openwrt/tmp/dl && rm -rf shadowsocksr-libev-2.5.6 && [ \! -d shadowsocksr-libev-2.5.6 ] && git clone  https://github.com/shadowsocksrr/shadowsocksr-libev shadowsocksr-libev-2.5.6 && (cd shadowsocksr-libev-2.5.6 && git checkout d63ff863800a5645aca4309d5dd5962bd1e95543 && git submodule update --init --recursive) && echo "Packing checkout..." && export TAR_TIMESTAMP=`cd shadowsocksr-libev-2.5.6 && git log -1 --format='\''@%ct'\''` && rm -rf shadowsocksr-libev-2.5.6/.git && 	tar --numeric-owner --owner=0 --group=0 --mode=a-s --sort=name ${TAR_TIMESTAMP:+--mtime="$TAR_TIMESTAMP"} -c shadowsocksr-libev-2.5.6 | 	xz -zc -7e > /home/runner/work/AutoBuild-Actions-Template/AutoBuild-Actions-Template/openwrt/tmp/dl/shadowsocksr-libev-2.5.6.tar.xz && mv /home/runner/work/AutoBuild-Actions-Template/AutoBuild-Actions-Template/openwrt/tmp/dl/shadowsocksr-libev-2.5.6.tar.xz /home/runner/work/AutoBuild-Actions-Template/AutoBuild-Actions-Template/openwrt/dl/ && rm -rf shadowsocksr-libev-2.5.6; );  )    '
mkdir -p /home/runner/work/AutoBuild-Actions-Template/AutoBuild-Actions-Template/openwrt/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-ipq40xx/stamp
SHELL= flock /home/runner/work/AutoBuild-Actions-Template/AutoBuild-Actions-Template/openwrt/tmp/.root-copy.flock -c 'cp -fpR /home/runner/work/AutoBuild-Actions-Template/AutoBuild-Actions-Template/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/shadowsocksr-libev-2.5.6/.pkgdir/shadowsocksr-libev-ssr-check/. /home/runner/work/AutoBuild-Actions-Template/AutoBuild-Actions-Template/openwrt/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-ipq40xx/'
touch /home/runner/work/AutoBuild-Actions-Template/AutoBuild-Actions-Template/openwrt/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-ipq40xx/stamp/.shadowsocksr-libev-ssr-check_installed
echo "shadowsocksr-libev-ssr-check" >> /home/runner/work/AutoBuild-Actions-Template/AutoBuild-Actions-Template/openwrt/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/pkginfo/shadowsocksr-libev.default.install
mkdir -p /home/runner/work/AutoBuild-Actions-Template/AutoBuild-Actions-Template/openwrt/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-ipq40xx/stamp
SHELL= flock /home/runner/work/AutoBuild-Actions-Template/AutoBuild-Actions-Template/openwrt/tmp/.root-copy.flock -c 'cp -fpR /home/runner/work/AutoBuild-Actions-Template/AutoBuild-Actions-Template/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/shadowsocksr-libev-2.5.6/.pkgdir/shadowsocksr-libev-ssr-local/. /home/runner/work/AutoBuild-Actions-Template/AutoBuild-Actions-Template/openwrt/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-ipq40xx/'
touch /home/runner/work/AutoBuild-Actions-Template/AutoBuild-Actions-Template/openwrt/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-ipq40xx/stamp/.shadowsocksr-libev-ssr-local_installed
echo "shadowsocksr-libev-ssr-local" >> /home/runner/work/AutoBuild-Actions-Template/AutoBuild-Actions-Template/openwrt/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/pkginfo/shadowsocksr-libev.default.install
mkdir -p /home/runner/work/AutoBuild-Actions-Template/AutoBuild-Actions-Template/openwrt/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-ipq40xx/stamp
SHELL= flock /home/runner/work/AutoBuild-Actions-Template/AutoBuild-Actions-Template/openwrt/tmp/.root-copy.flock -c 'cp -fpR /home/runner/work/AutoBuild-Actions-Template/AutoBuild-Actions-Template/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/shadowsocksr-libev-2.5.6/.pkgdir/shadowsocksr-libev-ssr-redir/. /home/runner/work/AutoBuild-Actions-Template/AutoBuild-Actions-Template/openwrt/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-ipq40xx/'
touch /home/runner/work/AutoBuild-Actions-Template/AutoBuild-Actions-Template/openwrt/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-ipq40xx/stamp/.shadowsocksr-libev-ssr-redir_installed
echo "shadowsocksr-libev-ssr-redir" >> /home/runner/work/AutoBuild-Actions-Template/AutoBuild-Actions-Template/openwrt/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/pkginfo/shadowsocksr-libev.default.install
mkdir -p /home/runner/work/AutoBuild-Actions-Template/AutoBuild-Actions-Template/openwrt/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-ipq40xx/stamp
SHELL= flock /home/runner/work/AutoBuild-Actions-Template/AutoBuild-Actions-Template/openwrt/tmp/.root-copy.flock -c 'cp -fpR /home/runner/work/AutoBuild-Actions-Template/AutoBuild-Actions-Template/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/shadowsocksr-libev-2.5.6/.pkgdir/shadowsocksr-libev-ssr-server/. /home/runner/work/AutoBuild-Actions-Template/AutoBuild-Actions-Template/openwrt/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-ipq40xx/'
[... cut ...]
make[2]: Leaving directory '/home/runner/work/AutoBuild-Actions-Template/AutoBuild-Actions-Template/openwrt'
make[1]: *** [package/Makefile:111: /home/runner/work/AutoBuild-Actions-Template/AutoBuild-Actions-Template/openwrt/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/stamp/.package_install] Error 2
make[1]: Leaving directory '/home/runner/work/AutoBuild-Actions-Template/AutoBuild-Actions-Template/openwrt'
make: *** [/home/runner/work/AutoBuild-Actions-Template/AutoBuild-Actions-Template/openwrt/include/toplevel.mk:230: world] Error 2
Error: Process completed with exit code 2.

Hyy2001X, 2021-07-01:
check_data_file_clashes: Package kmod-ath10k-ct-smallbuffers wants to install file /home/runner/work/AutoBuild-Actions-Template/AutoBuild-Actions-Template/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-ipq40xx/lib/modules/5.4.128/ath10k_core.ko
But that file is already provided by package * kmod-ath10k-ct
check_data_file_clashes: Package kmod-ath10k-ct-smallbuffers wants to install file /home/runner/work/AutoBuild-Actions-Template/AutoBuild-Actions-Template/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-ipq40xx/lib/modules/5.4.128/ath10k_pci.ko
But that file is already provided by package * kmod-ath10k-ct

检查一下这两个包 `kmod-ath10k-ct-smallbuffers` `kmod-ath10k-ct`

Hyy2001X, 2021-07-02:
上游 `ipq40xx: fix ath10k-ct conflit with ac58u`

closed 2021-07-02 by Hyy2001X