Hyy2001X/AutoBuild-Actions #88: 不支持所上传的映像文件格式，请选择适合当前平台的通用映像文件。

Paris0-0, 2021-07-15:
这两天编译出来的固件都不能直接刷,显示不支持所上传的映像文件格式，请选择适合当前平台的通用映像文件,进breed才能刷,请教下是为啥,newifi d2

Paris0-0, 2021-07-15:
看了下日志提示config无法保留,但是即使是刷同一个版本的固件还是报这个问题😂

Hyy2001X, 2021-07-15:
试试在终端里面刷，加一个参数 -F

NaiHeKK, 2021-07-16:
> 试试在终端里面刷，加一个参数 `-F` 应该是内核升级了 5.10 的问题

我的也这样，我还以为是我改的原因。升级不成功，到breed里可以刷入，但是刷入之后会开不了机

NaiHeKK, 2021-07-16:
> 这两天编译出来的固件都不能直接刷,显示不支持所上传的映像文件格式，请选择适合当前平台的通用映像文件,进breed才能刷,请教下是为啥,newifi d2

你breed刷入新的能开机吗

Hyy2001X, 2021-07-16:
> > 这两天编译出来的固件都不能直接刷,显示不支持所上传的映像文件格式，请选择适合当前平台的通用映像文件,进breed才能刷,请教下是为啥,newifi d2
> 
> 你breed刷入新的能开机吗

我开不了，但是有恩山网友刷了我的固件之后开机了

Paris0-0, 2021-07-16:
我14号编译的开机了，15号的还没试，每次都要重新配置太烦了。而且argon的配置程序也不见了

NaiHeKK, 2021-07-16:
> > > 这两天编译出来的固件都不能直接刷,显示不支持所上传的映像文件格式，请选择适合当前平台的通用映像文件,进breed才能刷,请教下是为啥,newifi d2
> > 
> > 
> > 你breed刷入新的能开机吗
> 
> 我开不了，但是有恩山网友刷了我的固件之后开机了

也是新三的吗？这不会也是啥玄学问题吧

Paris0-0, 2021-07-16:
新3，但配置我用的自己的

NaiHeKK, 2021-07-16:
> 我14号编译的开机了，15号的还没试，每次都要重新配置太烦了。而且argon的配置程序也不见了

我的是直接开不了机，好几个版本都这样的

NaiHeKK, 2021-07-16:
> 新3，但配置我用的自己的
看了下似乎只是精简了不少插件，难道是某个插件导致的吗

Hyy2001X, 2021-07-16:
> 我14号编译的开机了，15号的还没试，每次都要重新配置太烦了。而且argon的配置程序也不见了

可以备份一下配置 刷完以后再恢复

Paris0-0, 2021-07-16:
旧的配置恢复不上去，恢复了lan口就无法连接

Hyy2001X, 2021-07-16:
> 旧的配置恢复不上去，恢复了lan口就无法连接

5.4的备份不能用在5.10上了,5.10应该可以通用吧

Paris0-0, 2021-07-16:
> > 旧的配置恢复不上去，恢复了lan口就无法连接
> 
> 5.4的备份不能用在5.10上了,5.10应该可以通用吧

lean的源码不是一直5.4嘛😂

Paris0-0, 2021-07-16:
哦变成5.10了

Paris0-0, 2021-07-16:
昨天刷了最新编译的5.10，发现argon主题没了，选了不起作用，然后ssr plus 服务端ss协议也没了

wmyfelix, 2021-07-20:
> > > > 这两天编译出来的固件都不能直接刷,显示不支持所上传的映像文件格式，请选择适合当前平台的通用映像文件,进breed才能刷,请教下是为啥,newifi d2
> > > 
> > > 
> > > 你breed刷入新的能开机吗
> > 
> > 
> > 我开不了，但是有恩山网友刷了我的固件之后开机了
> 
> 也是新三的吗？这不会也是啥玄学问题吧

[#issuecomment-882347640](https://github.com/coolsnowwolf/lede/issues/7431#issuecomment-882347640)

closed 2021-08-06 by Hyy2001X