Hyy2001X/AutoBuild-Actions #100: 請問大神,如果要用官方源码编译,要改那些地方

pig1209, 2021-08-29:
我在x86_64.yml 改了這個地方
# 源码仓库:分支
  DEFAULT_SOURCE: openwrt/openwrt:openwrt-21.02
但编译错误

make: *** [/home/runner/work/AutoBuild-Actions/AutoBuild-Actions/openwrt/include/toplevel.mk:230: world] Error 2
Error: Process completed with exit code 2.

請問還有那些要改,謝謝

Hyy2001X, 2021-08-29:
`2021-08-29T08:50:29.6540095Z  * check_data_file_clashes: Package dnsmasq-full wants to install file /home/runner/work/AutoBuild-Actions/AutoBuild-Actions/openwrt/build_dir/target-x86_64_musl/root-x86/etc/hotplug.d/ntp/25-dnsmasqsec
2021-08-29T08:50:29.6541609Z 	But that file is already provided by package  * dnsmasq`
`2021-08-29T08:50:29.6534646Z  * check_data_file_clashes: Package libustream-openssl20201210 wants to install file /home/runner/work/AutoBuild-Actions/AutoBuild-Actions/openwrt/build_dir/target-x86_64_musl/root-x86/lib/libustream-ssl.so
2021-08-29T08:50:29.6536627Z 	But that file is already provided by package  * libustream-wolfssl20201210`

建议使用非`coolsnowwolf/lede`源码编译之前先设置`Scripts/AutoBuild_DiyScript.sh`文件中的`Load_Common_Config 和 Load_CustomPackages_List`这两行的值为`false`
简单的看了一下你的编译日志，第一个问题是 dnsmasq 与 dnsmasq-full 冲突了，解决方法是修改 .config 文件,添加如下两行
`CONFIG_PACKAGE_dnsmasq-full=y`
`# CONFIG_PACKAGE_dnsmasq is not set`
还有一个包 openssl 与默认的 wolfssl 冲突了，在 .config 底下加上 `# CONFIG_PACKAGE_luci-ssl-openssl is not set` 应该就可以了

pig1209, 2021-08-30:
謝謝,编译成功,但進入系統192.168.1.1 ,變成
/usr/lib/lua/luci/dispatcher.lua:725: No valid theme found
stack traceback:
	[C]: in function 'assert'
	/usr/lib/lua/luci/dispatcher.lua:725: in function 'init_template_engine'
	/usr/lib/lua/luci/dispatcher.lua:849: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:479: in function </usr/lib/lua/luci/dispatcher.lua:478>

Hyy2001X, 2021-08-30:
> 謝謝,编译成功,但進入系統192.168.1.1 ,變成
> /usr/lib/lua/luci/dispatcher.lua:725: No valid theme found
> stack traceback:
> [C]: in function 'assert'
> /usr/lib/lua/luci/dispatcher.lua:725: in function [...]

192.168.1.1:7681进入ttyd终端，然后试试指令 opkg remove luci-theme-argon，然后重启

Hyy2001X, 2021-08-30:
建议在本地搭建编译环境,自定义好.config文件再替换仓库里的.config,用仓库里默认的.config可能多多少少会有一些问题

pig1209, 2021-08-30:
大神,我以前用其他人成功的版本,在進入系統之前,都會有一些載入的動作,大概要1分鐘才會進入系統
但這個版本,開機約5秒就到 Booting the kernel  , Please press enter to activate this console,
然後按Eenter ,瞬間就到"Powered by Autobulild -Actions:...." 然後在brower 打192.168.1.1
就出現
/usr/lib/lua/luci/dispatcher.lua:725: No valid theme found
stack traceback:
[C]: in function 'assert'
/usr/lib/lua/luci/dispatcher.lua:725: in function 'init_template_engine'
/usr/lib/lua/luci/dispatcher.lua:849: in function 'dispatch'
/usr/lib/lua/luci/dispatcher.lua:479: in function </usr/lib/lua/luci/dispatcher.lua:478>
這是不是沒有載入什麼東西造成?

Hyy2001X, 2021-08-30:
> 大神,我以前用其他人成功的版本,在進入系統之前,都會有一些載入的動作,大概要1分鐘才會進入系統
> 但這個版本,開機約5秒就到 Booting the kernel , Please press enter to activate this console,
> 然後按Eenter ,瞬間就到"Powered by Autobulild -Actions:...." 然後在brower [...]

执行一下 opkg list | grep theme 返回的值我看看

pig1209, 2021-08-30:
沒有返回值

Hyy2001X, 2021-08-30:
> 沒有返回值

那就是没有主题包,编译的时候要带上 luci-theme-bootstrap
手动安装 opkg install luci-theme-bootstrap

pig1209, 2021-08-30:
我重新编译看看,謝謝

nnssz, 2021-09-22:
跟着学习官方原生openwrt21.02编译passwall ssrplus方法，🙏

closed 2021-11-02 by Hyy2001X