Hyy2001X/AutoBuild-Actions #108: 小娱百兆以及界面不能显示

7ack, 2021-09-24:
网口只有百兆
界面数据不能获取
[image]

Hyy2001X, 2021-09-25:
原因不明 试试更新到最新版的固件

7ack, 2021-09-25:
> 原因不明 试试更新到最新版的固件

改了lan地址就不能正常使用，改回1.1就好了，但是我的光猫也是1.1居然没冲突也是奇怪，还有小娱breed刷这个固件需要刷底包吗

Hyy2001X, 2021-09-25:
> > 原因不明 试试更新到最新版的固件
> 
> 改了lan地址就不能正常使用，改回1.1就好了，但是我的光猫也是1.1居然没冲突也是奇怪，还有小娱breed刷这个固件需要刷底包吗

和光猫的ip地址一样但是不冲突正常的 缺点就是不能访问光猫了 小娱应该和新三一样不需要刷底包

Hyy2001X, 2021-09-25:
> > 原因不明 试试更新到最新版的固件
> 
> 改了lan地址就不能正常使用，改回1.1就好了，但是我的光猫也是1.1居然没冲突也是奇怪，还有小娱breed刷这个固件需要刷底包吗

改完 lan ip 重启一下看看能不能恢复

7ack, 2021-09-25:
> > > 原因不明 试试更新到最新版的固件
> > 
> > 
> > 改了lan地址就不能正常使用，改回1.1就好了，但是我的光猫也是1.1居然没冲突也是奇怪，还有小娱breed刷这个固件需要刷底包吗
> 
> 改完 lan ip 重启一下看看能不能恢复

breed公版还有op都试了，刷了两遍都是这个情况，只能1.1下面正常使用，千兆降百兆我倒是没怎么注意，这几天关注一下再反馈

closed 2021-11-02 by Hyy2001X