Hyy2001X/AutoBuild-Actions #116: AutoUpdate下载版本错误

tossp, 2021-11-12:
应该下载后缀为`img.gz`的固件，获取到的名称却是`vmdk`后缀，是我哪里没配置对吗？

```
[23:59:06] [17283] [Quiet Mode] 正在下载,请耐心等待 ...
[23:59:06] [17283] 执行下载: [/usr/bin/wget --quiet --no-check-certificate --no-dns-cache -x -4 --tries 1 --timeout 5 -O /tmp/AutoUpdate.sh https://ghproxy.com/https://raw.githubusercontent.com/tossp/op/master/Scripts/AutoUpdate.sh]
[23:59:07] [17283] [Quiet Mode] 下载成功!
[23:59:07] [17283] 已删除文件: [/tmp/AutoUpdate.sh]
[23:59:11] [17542] [DOWNLOADER] Enabled No-Url-Filename Mode
[23:59:11] [17542] URL 数量: [1] 总重试次数: [1]
[23:59:11] [17542] 当前 URL: [https://ghproxy.com/https://raw.githubusercontent.com/tossp/op/master/Scripts/AutoUpdate.sh] URL 重试次数: [1]
[23:59:12] [17542] 正在下载脚本,请耐心等待 ...
[23:59:12] [17542] 已删除文件: [/tmp/AutoUpdate.sh]
[23:59:12] [17542] 执行下载: [/usr/bin/wget --quiet --no-check-certificate --no-dns-cache -x -4 --tries 1 --timeout 5 -O /tmp/AutoUpdate.sh https://ghproxy.com/https://raw.githubusercontent.com/tossp/op/master/Scripts/AutoUpdate.sh]
[23:59:14] [17542] 脚本下载成功!
[23:59:14] [17542] 脚本保存路径: [/bin]
[23:59:14] [17542] [V6.6.9 > V6.7.2] AutoUpdate 脚本更新成功!
[23:59:29] [17959] 固件更新指令: [sysupgrade -q]
[23:59:29] [17959] 特殊指令: [定时更新] / sysupgrade -q
[23:59:29] [17959] 执行: 更新固件
[23:59:31] [17959] Google 连接检查结果: [301]
[23:59:31] [17959] 正在检查版本更新 ...
[23:59:31] [17959] [CHECK_TIME] 文件: [/tmp/AutoUpdate/API] 验证失败!
[23:59:31] [17959] 已删除文件: [/tmp/AutoUpdate/API]
[23:59:31] [17959] URL 数量: [3] 总重试次数: [3]
[23:59:31] [17959] [DOWNLOADER] Enabled No-Url-Filename Mode
[23:59:31] [17959] 当前 URL: [https://ghproxy.com/https://github.com/tossp/op/releases/download/AutoUpdate/API] URL 重试次数: [1]
[23:59:32] [17959] [Quiet Mode] 正在下载固件信息,请耐心等待 ...
[23:59:32] [17959] 执行下载: [/usr/bin/wget --quiet --no-check-certificate -x -4 --tries 1 --timeout 5 -O /tmp/AutoUpdate/API_Cache https://ghproxy.com/https://github.com/tossp/op/releases/download/AutoUpdate/API]
[23:59:34] [17959] [Quiet Mode] 固件信息下载成功!
[23:59:34] [17959] 已删除文件: [/tmp/AutoUpdate/API]
[23:59:34] [17959] [ANALYZE_API] 开始解析 Github API ...
[23:59:38] [17959] [GET_FW_INFO] 获取固件名称: AutoBuild-lede-x86_64-R21.11.11-202111110801-UEFI-ff99a.vmdk
[23:59:38] [17959] [GET_FW_INFO] 获取下载次数: 4
[23:59:38] [17959] [GET_FW_INFO] 获取 SHA256 : ff99a
[23:59:38] [17959] [GET_FW_INFO] 获取固件版本: R21.11.11-202111110801
[23:59:38] [17959] [GET_FW_INFO] 获取发布日期: 20211111090729
[23:59:38] [17959] [GET_FW_INFO] 获取固件体积: 55.88MB
[23:59:38] [17959] [GET_FW_INFO] 获取固件链接: https://github.com/tossp/op/releases/download/AutoUpdate/AutoBuild-lede-x86_64-R21.11.11-202111110801-UEFI-ff99a.vmdk
[23:59:38] [17959] 当前版本: R21.11.11-202111090801
[23:59:38] [17959] 云端版本: R21.11.11-202111110801
[23:59:38] [17959] 云端固件名称: AutoBuild-lede-x86_64-R21.11.11-202111110801-UEFI-ff99a.vmdk
[23:59:38] [17959] 云端固件体积: 55.88MB
[23:59:38] [17959] 固件下载人数: 4
[23:59:38] [17959] [GET_CLOUD_LOG] 未检测到云端日志文件!
[23:59:38] [17959] [DOWNLOADER] Enabled No-Url-Filename Mode
[23:59:38] [17959] URL 数量: [3] 总重试次数: [5]
[23:59:38] [17959] 当前 URL: [https://github.com/tossp/op/releases/download/AutoUpdate/AutoBuild-lede-x86_64-R21.11.11-202111110801-UEFI-ff99a.vmdk] URL 重试次数: [2]
[23:59:39] [17959] 正在下载固件,请耐心等待 ...
[23:59:39] [17959] 已删除文件: [/tmp/AutoUpdate/AutoBuild-lede-x86_64-R21.11.11-202111110801-UEFI-ff99a.vmdk]
[23:59:39] [17959] 执行下载: [/usr/bin/wget --quiet --no-check-certificate -x -4 --tries 1 --timeout 5 -O /tmp/AutoUpdate/AutoBuild-lede-x86_64-R21.11.11-202111110801-UEFI-ff99a.vmdk https://github.com/tossp/op/releases/download/AutoUpdate/AutoBuild-lede-x86_64-R21.11.11-202111110801-UEFI-ff99a.vmdk]
[23:59:47] [17959] 固件下载成功!
[23:59:47] [17959] [GET_SHA256SUM] 计算结果: [ff99a]
[23:59:47] [17959] 固件 SHA256 比对通过!
[23:59:47] [17959] 卸载软件包 [gzip] ...
[23:59:47] [17959] 警告: 固件更新期间请不要断开电源或重启设备!
[23:59:50] [17959] 正在更新固件,请耐心等待 ...
[23:59:50] [17959] 固件刷入失败,请尝试使用 [autoupdate -F] 指令更新固件!
```

Hyy2001X, 2021-11-13:
感谢反馈，应该是脚本的问题，我晚上回去修复一下

Hyy2001X, 2021-11-13:
试试用最新的脚本来更新
`autoupdate -x --url https://raw.githubusercontent.com/Hyy2001X/AutoBuild-Actions/master/Scrip
ts/AutoUpdate.sh`

tossp, 2021-11-14:
谢谢，测试可用

closed 2021-11-14 by tossp

tossp, 2021-11-18:
/close 谢谢更新，可用了 从 手机 MailPlus 发送