Hyy2001X/AutoBuild-Actions #117: 是否删除了Argon主题插件？

wlfairlong, 2021-11-15:
你好，感谢您制作的固件，很好用！
我是NEWIFI-D2用户，使用10月28日的固件，该版本有Argon主题插件。自动更新到11月11日的固件时该插件就没有了，同时看到增加了广告过滤大师的插件。请问Argon主题插件能否加入回来呢？另外固件的更新能否提供一下更新日志，尤其对于增减插件强调一下。

Hyy2001X, 2021-11-15:
感谢反馈，没有删除这个主题，具体是哪里的问题我得看一看

wlfairlong, 2021-11-16:
好人一生平安

Hyy2001X, 2021-11-16:
R21.11.11-20211116 版本已经加回了 Argon 主题

wlfairlong, 2021-11-17:
从1028手动更新为1116固件后发现一些问题，下面是我操作过程和记录。

【自定义配置】
修改内网IP为10.10.10.1
修改WIFI名称和密码
添加SSR服务器并正常启用
添加一个Argon背景文件（1028固件默认可用为4.82MB，添加一张图片后可用4.7MB）
通过5G信号连接路由后功能一切正常，ping路由测试稳定

手动点击检测更新，发现11.16固件，点击更新，路由自动更新后：
1、网络不稳定，ping路由地址丢包超时频繁，大概半分钟超时一次（所有情况都是5G WIFI的无线连接状态下测试）
2、登陆界面被初始化，Argon背景文件列表被清空，可用空间变小（升级为1116后可用空间显示仅有3.29MB）

尝试清空配置，直接刷1116固件，恢复上述“自定义配置”设置，没有发现不稳定的现象，Argon默认可用空间4.3MB

尝试清空配置，直接刷1028固件，恢复上述“自定义配置”设置，再次手动更新为1116，上述问题再次复现。

插件我只使用了SSR和Argon，其他的功能没测试

Hyy2001X, 2021-11-17:
> 从1028手动更新为1116固件后发现一些问题，下面是我操作过程和记录。
>
> 【自定义配置】 修改内网IP为10.10.10.1 修改WIFI名称和密码 添加SSR服务器并正常启用 添加一个Argon背景文件（1028固件默认可用为4.82MB，添加一张图片后可用4.7MB） 通过5G信号连接路由后功能一切正常，ping路由测试稳定 [...]

试过更新完固件后再手动重启一遍吗，情况是不是还是如此

wlfairlong, 2021-11-18:
第一次升级完后发现不稳定的问题后重启过一次，网络不稳定的问题没有改善，后面就没有在刷完后单独重启过

NaiHeKK, 2021-11-18:
我就不会有这问题了，升级完直接重置了。(T^T)

wlfairlong, 2021-11-18:
> 我就不会有这问题了，升级完直接重置了。(T^T)

我第二次刷回1028再升上1116的时候，自定义过的10.10.10.1变回了192.168.1.1，我不确定跟我的操作有没有关系，所以上面没写

NaiHeKK, 2021-11-19:
> > 我就不会有这问题了，升级完直接重置了。(T^T)
> 
> 我第二次刷回1028再升上1116的时候，自定义过的10.10.10.1变回了192.168.1.1，我不确定跟我的操作有没有关系，所以上面没写

有可能空间不足然后配置丢失了，自定义ip还是在编译前改比较好

NaiHeKK, 2021-11-19:
> > 从1028手动更新为1116固件后发现一些问题，下面是我操作过程和记录。
> > 【自定义配置】 修改内网IP为10.10.10.1 修改WIFI名称和密码 添加SSR服务器并正常启用 添加一个Argon背景文件（1028固件默认可用为4.82MB，添加一张图片后可用4.7MB） 通过5G信号连接路由后功能一切正常，ping路由测试稳定
> > 手动点击检测更新，发现11.16固件，点击更新，路由自动更新后： 1、网络不稳定，ping路由地址丢包超时频繁，大概半分钟超时一次（所有情况都是5G WIFI的无线连接状态下测试） 2、登陆界面被初始化，Argon背景文件列表被清空，可用空间变小（升级为1116后可用空间显示仅有3.29MB）
> > 尝试清空配置，直接刷1116固件，恢复上述“自定义配置”设置，没有发现不稳定的现象，Argon默认可用空间4.3MB
> > 尝试清空配置，直接刷1028固件，恢复上述“自定义配置”设置，再次手动更新为1116，上述问题再次复现。
> > 插件我只使用了SSR和Argon，其他的功能没测试
> 
> 试过更新完固件后再手动重启一遍吗，情况是不是还是如此 最近版本改动的东西挺多的，可能还是这个原因导致的

大佬，有没有什么办法让我挂载的U盘更新后挂载点不变啊

Hyy2001X, 2021-11-23:
> 大佬，有没有什么办法让我挂载的U盘更新后挂载点不变啊

系统-挂载点，手动添加挂载点就好了，选 UUID 匹配

closed 2021-11-29 by Hyy2001X