Hyy2001X/AutoBuild-Actions #119: 设置拨号时提示未安装支持协议PPP-MOD-PPPOE但是可以拨号

smdx023, 2021-11-16:
新3 使用的release里面的版本 AutoBuild-lede-d-team_newifi-d2-R21.11.11-20211116-5c879.bin

https://github.com/smdx023/AutoBuild-Actions/blob/master/debug1.png

smdx023, 2021-11-16:
dnspod里面设置完全正确，最后报错域名错误。不知道是插件除了问题还是安装过程中存在一些bug

 021535       : parsing script '/usr/lib/ddns/update_dnspod_cn.sh'
 021536       : #> wget --no-hsts -nv -t 1 -O /var/run/ddns/dnspod.dat -o /var/run/ddns/dnspod.err --no-proxy --post-data 'login_token=271830,***PW***&format=json&domain=90263675&sub_domain=@&record_type=AAAA' https://dnsapi.cn/Record.List
 021539 ERROR : Error message:[域名不正确，请输入主域名，如 dnspod.cn]
 021540  WARN : Transfer failed - 1 Try again in 60 seconds

Hyy2001X, 2021-11-17:
上游更新 ppp 的问题，下个版本会修复

closed 2021-11-23 by Hyy2001X