Hyy2001X/AutoBuild-Actions #121: 编译错误

tossp, 2021-11-19:
没太看懂这个日志，像是`openwrt-passwall`的某个依赖没下载到

感觉上，是上游依赖openwrt-passwall的问题，但还是反馈一下

```
2021-11-19T08:51:15.8466337Z make[3]: Entering directory '/home/runner/work/op/op/openwrt/package/other/openwrt-passwall/naiveproxy'
2021-11-19T08:51:16.0017566Z mkdir -p /home/runner/work/op/op/openwrt/dl
2021-11-19T08:51:16.0052119Z SHELL= flock /home/runner/work/op/op/openwrt/tmp/.chrome-linux-4664-1636557077-6e390f4e505916531ca2ab0c895d5903ab4d88a9.profdata.flock -c '  	/home/runner/work/op/op/openwrt/scripts/download.pl "/home/runner/work/op/op/openwrt/dl" "chrome-linux-4664-1636557077-6e390f4e505916531ca2ab0c895d5903ab4d88a9.profdata" "8dcf5973033d40c9a7b15e571dea3832e7b67976aad9113369e22d43808c603f" "chrome-linux-4664-1636557077-6e390f4e505916531ca2ab0c895d5903ab4d88a9.profdata" "https://storage.googleapis.com.cnpmjs.org/chromium-optimization-profiles/pgo_profiles" "https://storage.googleapis.com/chromium-optimization-profiles/pgo_profiles"    '
2021-11-19T08:51:16.0360702Z + curl -f --connect-timeout 20 --retry 5 --location --insecure https://storage.googleapis.com.cnpmjs.org/chromium-optimization-profiles/pgo_profiles/chrome-linux-4664-1636557077-6e390f4e505916531ca2ab0c895d5903ab4d88a9.profdata
2021-11-19T08:51:16.0442453Z   % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
2021-11-19T08:51:16.0443488Z                                  Dload  Upload   Total   Spent    Left  Speed
2021-11-19T08:51:16.0444089Z 
2021-11-19T08:51:16.8928958Z   0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
2021-11-19T08:51:17.8387105Z   0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
2021-11-19T08:51:18.9266495Z   0 47.5M    0 48240    0     0  26889      0  0:30:55  0:00:01  0:30:54 26874
2021-11-19T08:51:19.7964382Z   2 47.5M    2 1263k    0     0   438k      0  0:01:51  0:00:02  0:01:49  438k
2021-11-19T08:51:20.8844794Z  21 47.5M   21 10.1M    0     0  2767k      0  0:00:17  0:00:03  0:00:14 2766k
2021-11-19T08:51:21.7963650Z  48 47.5M   48 23.2M    0     0  4918k      0  0:00:09  0:00:04  0:00:05 4917k
2021-11-19T08:51:22.7103981Z  73 47.5M   73 34.8M    0     0  6208k      0  0:00:07  0:00:05  0:00:02 7282k
2021-11-19T08:51:22.7105552Z 100 47.5M  100 47.5M    0     0  7310k      0  0:00:06  0:00:06 --:--:-- 9995k
2021-11-19T08:51:22.7185831Z Hash of the downloaded file does not match (file: 5f693c4779c1d6802e509c2c1e5f130017fd6634a0ccd3a0de47be419c068f5e, requested: 8dcf5973033d40c9a7b15e571dea3832e7b67976aad9113369e22d43808c603f) - deleting download.
2021-11-19T08:51:22.7350181Z + curl -f --connect-timeout 20 --retry 5 --location --insecure https://storage.googleapis.com/chromium-optimization-profiles/pgo_profiles/chrome-linux-4664-1636557077-6e390f4e505916531ca2ab0c895d5903ab4d88a9.profdata
2021-11-19T08:51:22.7422088Z   % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
2021-11-19T08:51:22.7423100Z                                  Dload  Upload   Total   Spent    Left  Speed
2021-11-19T08:51:22.7423698Z 
2021-11-19T08:51:22.8500044Z   0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
2021-11-19T08:51:23.3048897Z   0 47.5M    0 12554    0     0   114k      0  0:07:05 --:--:--  0:07:05  113k
2021-11-19T08:51:23.3049813Z 100 47.5M  100 47.5M    0     0  84.9M      0 --:--:-- --:--:-- --:--:-- 84.8M
2021-11-19T08:51:23.3114194Z Hash of the downloaded file does not match (file: 5f693c4779c1d6802e509c2c1e5f130017fd6634a0ccd3a0de47be419c068f5e, requested: 8dcf5973033d40c9a7b15e571dea3832e7b67976aad9113369e22d43808c603f) - deleting download.
2021-11-19T08:51:23.3281454Z + curl -f --connect-timeout 20 --retry 5 --location --insecure https://sources.cdn.openwrt.org/chrome-linux-4664-1636557077-6e390f4e505916531ca2ab0c895d5903ab4d88a9.profdata
2021-11-19T08:51:23.3362454Z   % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
2021-11-19T08:51:23.3363428Z                                  Dload  Upload   Total   Spent    Left  Speed
2021-11-19T08:51:23.3364014Z 
2021-11-19T08:51:23.5810883Z   0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
2021-11-19T08:51:23.5812726Z   0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
2021-11-19T08:51:23.5813512Z curl: (22) The requested URL returned error: 404 
2021-11-19T08:51:23.5828823Z Download failed.
2021-11-19T08:51:23.5915529Z + curl -f --connect-timeout 20 --retry 5 --location --insecure https://sources.openwrt.org/chrome-linux-4664-1636557077-6e390f4e505916531ca2ab0c895d5903ab4d88a9.profdata
2021-11-19T08:51:23.5986775Z   % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
2021-11-19T08:51:23.5988041Z                                  Dload  Upload   Total   Spent    Left  Speed
2021-11-19T08:51:23.5988419Z 
2021-11-19T08:51:23.7838809Z   0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
2021-11-19T08:51:24.2226296Z   0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
2021-11-19T08:51:24.2228708Z   0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
2021-11-19T08:51:24.2237742Z curl: (22) The requested URL returned error: 404 Not Found
2021-11-19T08:51:24.2256383Z Download failed.
2021-11-19T08:51:24.2347367Z + curl -f --connect-timeout 20 --retry 5 --location --insecure https://mirror2.openwrt.org/sources/chrome-linux-4664-1636557077-6e390f4e505916531ca2ab0c895d5903ab4d88a9.profdata
2021-11-19T08:51:24.2419515Z   % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
2021-11-19T08:51:24.2420593Z                                  Dload  Upload   Total   Spent    Left  Speed
2021-11-19T08:51:24.2421138Z 
2021-11-19T08:51:24.7837636Z   0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
2021-11-19T08:51:24.9736790Z   0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
2021-11-19T08:51:24.9737487Z   0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
2021-11-19T08:51:24.9742590Z curl: (22) The requested URL returned error: 404 Not Found
2021-11-19T08:51:24.9758498Z Download failed.
2021-11-19T08:51:24.9759220Z No more mirrors to try - giving up.
2021-11-19T08:51:24.9771589Z make[3]: *** [Makefile:130: /home/runner/work/op/op/openwrt/dl/chrome-linux-4664-1636557077-6e390f4e505916531ca2ab0c895d5903ab4d88a9.profdata] Error 2
2021-11-19T08:51:24.9785434Z make[3]: Leaving directory '/home/runner/work/op/op/openwrt/package/other/openwrt-passwall/naiveproxy'
2021-11-19T08:51:24.9786874Z time: package/other/openwrt-passwall/naiveproxy/compile#0.96#0.78#9.13
2021-11-19T08:51:24.9787990Z     ERROR: package/other/openwrt-passwall/naiveproxy failed to build.
2021-11-19T08:51:24.9791799Z make[2]: *** [package/Makefile:116: package/other/openwrt-passwall/naiveproxy/compile] Error 1
2021-11-19T08:51:24.9796041Z make[2]: Leaving directory '/home/runner/work/op/op/openwrt'
2021-11-19T08:51:24.9813227Z make[1]: *** [package/Makefile:110: /home/runner/work/op/op/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
2021-11-19T08:51:24.9814273Z make[1]: Leaving directory '/home/runner/work/op/op/openwrt'
2021-11-19T08:51:24.9823067Z make: *** [/home/runner/work/op/op/openwrt/include/toplevel.mk:230: world] Error 2
2021-11-19T08:51:24.9832262Z ##[error]Process completed with exit code 2.
2021-11-19T08:51:25.0001367Z ##[group]Run GitRML/delete-workflow-runs@main
2021-11-19T08:51:25.0001884Z with:
2021-11-19T08:51:25.0002232Z   retain_days: 7
2021-11-19T08:51:25.0002614Z   keep_minimum_runs: 3
2021-11-19T08:51:25.0003409Z   token: ***
2021-11-19T08:51:25.0003779Z   repository: tossp/op
2021-11-19T08:51:25.0004157Z env:
2021-11-19T08:51:25.0004502Z   CONFIG_FILE: x86_64
2021-11-19T08:51:25.0004980Z   DEFAULT_SOURCE: coolsnowwolf/lede:master
2021-11-19T08:51:25.0005470Z   UPLOAD_RELEASES: true
2021-11-19T08:51:25.0005901Z   UPLOAD_ARTIFACTS: false
2021-11-19T08:51:25.0006350Z   UPLOAD_BIN_ARTIFACTS: true
2021-11-19T08:51:25.0006793Z   DELETE_USELESS_FILES: true
2021-11-19T08:51:25.0007217Z   DELETE_OLD_WORKFLOW: true
2021-11-19T08:51:25.0007800Z   REPO_URL: https://github.com/coolsnowwolf/lede
2021-11-19T08:51:25.0008346Z   REPO_BRANCH: master
2021-11-19T08:51:25.0008747Z   Artifacts_Date: 11190801
2021-11-19T08:51:25.0009147Z   TARGET_DEVICE_ARCH: 
2021-11-19T08:51:25.0009518Z ##[endgroup]
2021-11-19T08:51:25.8823045Z No workflow runs need to be deleted.
2021-11-19T08:51:25.8954886Z Post job cleanup.
2021-11-19T08:51:26.0144904Z [command]/usr/bin/git version
2021-11-19T08:51:26.0192685Z git version 2.33.1
```

NaiHeKK, 2021-11-25:
我也添加了passwall，昨晚编译通过。没问题的话请关闭Is

tossp, 2021-11-26:
抱歉，，，昨天的编译通过了，谢谢

closed 2021-11-26 by tossp