Hyy2001X/AutoBuild-Actions #123: 红米ac2100没有无线信号？

zhuzf88, 2021-11-29:
从别的openwrt刷机过来发现没有无线信号，重置也没用。

直接用breed刷需要过渡固件吗？

NaiHeKK, 2021-11-29:
不需要

zhuzf88, 2021-11-29:
> 不需要

breed提示：无法判断固件类型

NaiHeKK, 2021-11-29:
> > 不需要
> 
> breed提示：无法判断固件类型

请给出操作步骤，确认下载的是否是对应的固件

zhuzf88, 2021-11-29:
机型：红米ac2100
breed版本：r1286
固件： https://github.com/Hyy2001X/AutoBuild-Actions/releases/download/AutoUpdate/AutoBuild-lede-xiaomi_redmi-router-ac2100-R21.11.11-20211122-c8779.bin

Hyy2001X, 2021-11-29:
搜不到信号的话用 80MHz 频宽

Hyy2001X, 2021-11-29:
> 从别的openwrt刷机过来发现没有无线信号，重置也没用。
> 
> 直接用breed刷需要过渡固件吗？

先刷一个底包，然后再更新到现在的固件

zhuzf88, 2021-11-29:
> > 从别的openwrt刷机过来发现没有无线信号，重置也没用。
> > 直接用breed刷需要过渡固件吗？
> 
> 先刷一个底包，然后再更新到现在的固件

多谢，刷底包再刷就搞定了。估计是和我原来openwrt固件有冲突。

zhuzf88, 2021-11-29:
这固件是不是开源驱动？感觉信号明显比原来的差。

Hyy2001X, 2021-11-29:
> 这固件是不是开源驱动？感觉信号明显比原来的差。

是开源的驱动

zhuzf88, 2021-11-29:
> > 这固件是不是开源驱动？感觉信号明显比原来的差。
> 
> 是开源的驱动，感觉不好用可以Fork以后修改成闭源驱动编译，配置文件里标注了开源和闭源的内容
> 
> https://github.com/Hyy2001X/AutoBuild-Actions/blob/668bd2771e5f7cec3da2459b05fa8a6e25f7aa24/Configs/xiaomi_redmi-router-ac2100#L22

多谢了，非常好的固件。界面太漂亮了。

closed 2021-11-29 by Hyy2001X

NaiHeKK, 2021-11-30:
> > 从别的openwrt刷机过来发现没有无线信号，重置也没用。
> > 直接用breed刷需要过渡固件吗？
> 
> 先刷一个底包，然后再更新到现在的固件

还有底包这种东西吗？我都是直接breed里恢复出厂设置然后直接更新固件的啊

Hyy2001X, 2021-11-30:
> > > 从别的openwrt刷机过来发现没有无线信号，重置也没用。
> > > 直接用breed刷需要过渡固件吗？
> > 
> > 
> > 先刷一个底包，然后再更新到现在的固件
> 
> 还有底包这种东西吗？我都是直接breed里恢复出厂设置然后直接更新固件的啊

某些型号是需要刷底包的

NaiHeKK, 2021-11-30:
> > > > 从别的openwrt刷机过来发现没有无线信号，重置也没用。
> > > > 直接用breed刷需要过渡固件吗？
> > > 
> > > 
> > > 先刷一个底包，然后再更新到现在的固件
> > 
> > 
> > 还有底包这种东西吗？我都是直接breed里恢复出厂设置然后直接更新固件的啊
> 
> 某些型号是需要刷底包的

哦哦，谢谢

zhuzf88, 2021-11-30:
> > > 从别的openwrt刷机过来发现没有无线信号，重置也没用。
> > > 直接用breed刷需要过渡固件吗？
> > 
> > 
> > 先刷一个底包，然后再更新到现在的固件
> 
> 还有底包这种东西吗？我都是直接breed里恢复出厂设置然后直接更新固件的啊

小米的固件好像分区不一样，刷op都要先刷底包过渡

> > 这固件是不是开源驱动？感觉信号明显比原来的差。
> 
> 是开源的驱动，感觉不好用可以Fork以后修改成闭源驱动编译，配置文件里标注了开源和闭源的内容
> 
> https://github.com/Hyy2001X/AutoBuild-Actions/blob/668bd2771e5f7cec3da2459b05fa8a6e25f7aa24/Configs/xiaomi_redmi-router-ac2100#L22

我编译了闭源驱动固件，但是从luci的系统页面刷鞋固件总是页面挂死没反应，可以直接mtd刷写吗？

Hyy2001X, 2021-12-01:
> 

概况-释放内存，然后再更新固件试试