Hyy2001X/AutoBuild-Actions #145: 新3无法扩展USB空间

lbcz-tian, 2022-03-04:
用最新的0206版本，无法配置USB空间扩展😢😢，麻烦大佬看看是咋回事呀
[image]

acgbox233, 2022-03-09:
我的也是这个毛病。

lbcz-tian, 2022-03-09:
试了一圈，我这用2211215版本可以正常扩展空间，暂时不升级了😢

acgbox233, 2022-03-10:
> 试了一圈，我这用20211215版本可以正常扩展空间，暂时不升级了😢
我找不到旧版本了，作者这边把旧版本删了。

lbcz-tian, 2022-03-10:
> > 试了一圈，我这用20211215版本可以正常扩展空间，暂时不升级了😢
> 
> 我找不到旧版本了，作者这边把旧版本删了。 目前我换成了H大的老毛子padavan了。

padavan咋样，我没用过额
旧版在恩山论坛里有，提取码tw2s
https://pan.baidu.com/s/1T69H1nZ_y8RqMhc80nV8LQ

NaiHeKK, 2022-03-10:
我都是通过挂载点挂载到overlay的，就是每次升级完需要格一次U盘

acgbox233, 2022-03-10:
> > > 试了一圈，我这用20211215版本可以正常扩展空间，暂时不升级了😢
> > 
> > 
> > 我找不到旧版本了，作者这边把旧版本删了。 目前我换成了H大的老毛子padavan了。
> 
> padavan咋样，我没用过额 旧版在恩山论坛里有，提取码tw2s https://pan.baidu.com/s/1T69H1nZ_y8RqMhc80nV8LQ

最新编译的openwrt固件在我这台新路由3上容易卡死重启，插上U盘的一瞬间也会卡死并重启，以前的版本从没出现过这种现象。我还一度以为是路由器本身老化了或者硬件出故障了。而换回到padavan后则稳定了很多，WiFi质量也更好。目前看来没有遇到卡死又重启的现象。

附上padavan官方下载地址：[https://opt.cn2qq.com/padavan/](https://opt.cn2qq.com/padavan/)

lbcz-tian, 2022-03-10:
> 最新编译的openwrt固件在我这台新路由3上容易卡死重启，插上U盘的一瞬间也会卡死并重启，以前的版本从没出现过这种现象。我还一度以为是路由器本身老化了或者硬件出故障了。而换回到padavan后则稳定了很多，WiFi质量也更好。目前看来没有遇到卡死又重启的现象。
>
> 附上padavan官方下载地址：https://opt.cn2qq.com/padavan/

感谢老哥🤗
我自己fork后编译的固件，也是会不定时重启，用了4天重启了5次，我先看看旧版本会不会有这个问题😂，如果还这样我就刷padavan看看

lbcz-tian, 2022-03-10:
> 我都是通过挂载点挂载到overlay的，就是每次升级完需要格一次U盘

我搜搜教程，还不太会整🤣

acgbox233, 2022-03-10:
> > 我都是通过挂载点挂载到overlay的，就是每次升级完需要格一次U盘
> 
> 我搜搜教程，还不太会整🤣

我之前不知道这个固件有自动挂载功能，一直都是手动挂载的，参考的是这个教程：[https://iyzm.net/openwrt/92.html](https://iyzm.net/openwrt/92.html)，实测可行。

Hyy2001X, 2022-03-13:
> 用最新的0206版本，无法配置USB空间扩展😢😢，麻烦大佬看看是咋回事呀 [image]

感谢反馈，下个版本修复

Hyy2001X, 2022-03-13:
> 我的也是这个毛病。

感谢反馈，下个版本修复

lbcz-tian, 2022-03-13:
> > 用最新的0206版本，无法配置USB空间扩展😢😢，麻烦大佬看看是咋回事呀 [image]
> 
> 感谢反馈，下个版本修复

哇大佬要恢复更新了吗，期待一波😋😋

lbcz-tian, 2022-03-16:
新版0313已可扩展空间，感谢大佬们。
我之前用1215旧版会经常性重启，不知道啥原因，个人使用会挂PT，大约5000种，查看连接数大约在6000-9000波动。先换0313版看看会不会有重启，再次感谢大佬们😋😋

closed 2022-03-16 by lbcz-tian