Hyy2001X/AutoBuild-Actions #148: 添加passwall插件报错

lgfyxx, 2022-03-07:
请教：

修改了AutoBuild_DiyScript.sh ，给NEWIFI D2添加了passwall的插件，之前一直编译没有问题，最近又编译完一个X86的固件后，
NEWIFI 这边就开始报错，把整个库删了重新fork还是一样的错误：

添加代码如下（加粗部分）：

case "${TARGET_PROFILE}" in
		d-team_newifi-d2)
			**AddPackage git other openwrt-passwall xiaorouji main
			rm -rf packages/lean/autocore
			AddPackage git lean autocore-modify Hyy2001X master**
			patch < ${CustomFiles}/${TARGET_PROFILE}_mac80211.patch -p1 -d ${Home}
			Copy ${CustomFiles}/${TARGET_PROFILE}_system ${BASE_FILES}/etc/config system
			sed -i "/DEVICE_COMPAT_VERSION := 1.1/d" target/linux/ramips/image/mt7621.mk

最新报错如下：

[14:32:47] Checking out package [openwrt-passwall] to package/other ...
Error: Process completed with exit code 128.

非常感谢！

lbcz-tian, 2022-03-07:
我也是一样的问题，passwall好像把分支名改为`packages`了，我改了之后编译没报错，但是编译出来的固件里没有passwall。小白一个，我也不知道咋回事了😢

NaiHeKK, 2022-03-07:
> 我也是一样的问题，passwall好像把分支名改为`packages`了，我改了之后编译没报错，但是编译出来的固件里没有passwall。小白一个，我也不知道咋回事了😢

Config里也加了吗

lbcz-tian, 2022-03-07:
> 

加了的，我不知道我config配置有没有别的问题，大佬也可以去我仓库看下😢😢

lgfyxx, 2022-03-08:
> 我也是一样的问题，passwall好像把分支名改为`packages`了，我改了之后编译没报错，但是编译出来的固件里没有passwall。小白一个，我也不知道咋回事了😢

得知据说是特殊时期代码不开放，不太懂代码。。。

281677160, 2022-03-08:
passwall作者把他仓库里面的luci-app-passwall这个luci删除了，只保留了其他，你们只要找luci-app-passwall补全就可以用了
或者拉我补全的仓库也可以
https://github.com/281677160/openwrt-passwall

281677160, 2022-03-10:
passwall作者昨天弄了一个简化版的，有需要的去拉取使用吧
https://github.com/xiaorouji/openwrt-passwall2
https://github.com/xiaorouji/openwrt-passwall
把这样两个库拉上就行了，插件名称是luci-app-passwall2

lbcz-tian, 2022-03-10:
> passwall作者昨天弄了一个简化版的，有需要的去拉取使用吧 https://github.com/xiaorouji/openwrt-passwall2 https://github.com/xiaorouji/openwrt-passwall 把这样两个库拉上就行了，插件名称是luci-app-passwall2

我之前用老哥你的库编译报错了，我也不知道是啥错误😂就换旧版了
`Run chmod +x Scripts/AutoBuild_*.sh
  chmod +x Scripts/AutoBuild_*.sh
  cp $GITHUB_WORKSPACE/Configs/$CONFIG_FILE openwrt/.config
  cd openwrt
  make defconfig
  source $GITHUB_WORKSPACE/Scripts/AutoBuild_DiyScript.sh
  source $GITHUB_WORKSPACE/Scripts/AutoBuild_Function.sh
  Firmware_Diy_Before
  rm .config && cp $GITHUB_WORKSPACE/Configs/$CONFIG_FILE .config
  Firmware_Diy_Main
  Firmware_Diy
  shell: /usr/bin/bash -e {0}
  env:
    CONFIG_FILE: d-team_newifi-d[2](https://github.com/lbcz-tian/AutoBuild-Actions/runs/5458792391?check_suite_focus=true#step:7:2)-Passwall
    DEFAULT_SOURCE: coolsnowwolf/lede:master
    UPLOAD_RELEASES: true
    UPLOAD_ARTIFACTS: false
    UPLOAD_BIN_ARTIFACTS: false
    DELETE_USELESS_FILES: true
    DELETE_OLD_WORKFLOW: true
    Tempoary_FLAG: Passwall
    REPO_URL: https://github.com/coolsnowwolf/lede
    REPO_BRANCH: master
    Compile_Date: 20220[3](https://github.com/lbcz-tian/AutoBuild-Actions/runs/5458792391?check_suite_focus=true#step:7:3)08025[4](https://github.com/lbcz-tian/AutoBuild-Actions/runs/5458792391?check_suite_focus=true#step:7:4)
    Display_Date: 2022/03/08
/home/runner/work/AutoBuild-Actions/AutoBuild-Actions/openwrt/.config:94: *** missing separator.  Stop.
make: *** [/home/runner/work/AutoBuild-Actions/AutoBuild-Actions/openwrt/include/toplevel.mk:[7](https://github.com/lbcz-tian/AutoBuild-Actions/runs/5458792391?check_suite_focus=true#step:7:7)[9](https://github.com/lbcz-tian/AutoBuild-Actions/runs/5458792391?check_suite_focus=true#step:7:9): prepare-tmpinfo] Error 2
Error: Process completed with exit code 2.`

281677160, 2022-03-10:
> 

没看明白你的啥报错，呵呵，我自己备份的库我自己编译是没问题的，今天测试了作者的简化版，也是没问题的

lbcz-tian, 2022-03-10:
> > 
> 
> 没看明白你的啥报错，呵呵，我自己备份的库我自己编译是没问题的，今天测试了作者的简化版，也是没问题的

我也没看明白😆，我再试试

closed 2022-04-22 by Hyy2001X