Hyy2001X/AutoBuild-Actions #155: 这个如何添加自己要增加的源码仓库

luchen7, 2022-03-31:
这个如何添加自己要增加的源码仓库

Hyy2001X, 2022-04-01:
编辑 Scripts/AutoBuild_DiyScript.sh
如果要添加 passwall，就在23行到87之间加上这两行
AddPackage git passwall-depends openwrt-passwall xiaorouji packages
AddPackage git passwall-luci openwrt-passwall xiaorouji luci

luchen7, 2022-04-03:
NEWIFI D2 / Compile 
Failed in 5 minutes and 49 seconds
[image]

luchen7, 2022-04-03:
[image]

luchen7, 2022-04-03:
编译时候报错

Hyy2001X, 2022-04-03:
> 编译时候报错

位置加错了, 加到 esac 的下一行

luchen7, 2022-04-03:
我图片上的89和90中间吗

luchen7, 2022-04-03:
还是81和82中间

Hyy2001X, 2022-04-03:
> 我图片上的89和90中间吗

是的

luchen7, 2022-04-03:
好的 对了  passwall  xiaorouji好像弄成什么passwall2了  跟这个没关系拔

Hyy2001X, 2022-04-03:
> 好的 对了 passwall xiaorouji好像弄成什么passwall2了 跟这个没关系拔

openwrt-passwall2 仓库要这么写,有点区别: `AddPackage git passwall-luci openwrt-passwall2 xiaorouji main`

luchen7, 2022-04-03:
那刚才的两条还要增加吗  还是只增加这一条

luchen7, 2022-04-03:
给个建议啊  能否直接给passwall增加进去  多一个选择  你的这个固件太好用了  适合新手  我现在有16台设备再用你的固件  主要我需要PASSWALL的按端口分流

luchen7, 2022-04-03:
[image]
还是编译错误

Hyy2001X, 2022-04-03:
> 

你按照我的最新提交修改一下那一行

luchen7, 2022-04-03:
你最新提交的是不是已经带PASSWALL了？

luchen7, 2022-04-03:
可以加下TG吗？https://t.me/xygggggg

closed 2022-04-14 by Hyy2001X