Hyy2001X/AutoBuild-Actions #165: NEWIFI-D2 升级内核为5.10导致无法联网

NaiHeKK, 2022-04-23:
大佬，我尝试修改NEWIFI-D2的lede内核为5.10，编译成功了但是刷上之后就没网了。用网线直接连不上电脑，用无线是没网。大佬能不能帮忙看看啥情况啊。
  非常感谢！

NaiHeKK, 2022-07-11:
大佬，能不能帮忙研究下为啥啊，我看其它人编译的好像能联网的

NaiHeKK, 2022-07-12:
> 用默认的配置去编译也不行吗

这个还没试，现在在一点点的试。就是试一次编译要好长时间

NaiHeKK, 2022-07-12:
> 用默认的配置去编译也不行吗

https://github.com/kiddin9/OpenWrt_x86-r2s-r4s-r5s-N1这个仓库的应该是可以用的

Hyy2001X, 2022-07-12:
试试[这个](https://github.com/Hyy2001X/AutoBuild-Actions/releases/download/AutoUpdate/openwrt-ramips-mt7621-d-team_newifi-d2-squashfs-sysupgrade-5.10.bin)固件， 不保留配置刷入

登录信息
192.168.1.1 注意不能和光猫冲突
root
password

NaiHeKK, 2022-07-12:
> 试试[这个](https://github.com/Hyy2001X/AutoBuild-Actions/releases/download/AutoUpdate/openwrt-ramips-mt7621-d-team_newifi-d2-squashfs-sysupgrade-5.10.bin)固件， 不保留配置刷入
> 
> 登录信息 192.168.1.1 注意不能和光猫冲突 root password

好咧。

- et0.1失效了而已。重新编译吧，本地编译也只需要一个小时而已。我看了你提供的D2配置文件，太古老了，这段时间openwrt官方更新了很多MT7621的驱动，L更是直接把新版的闭源驱动直接整合了，建议考虑make clear之后重配置。
这是我在恩山提问有人回复的不知道有没有用

NaiHeKK, 2022-07-12:
> 试试[这个](https://github.com/Hyy2001X/AutoBuild-Actions/releases/download/AutoUpdate/openwrt-ramips-mt7621-d-team_newifi-d2-squashfs-sysupgrade-5.10.bin)固件， 不保留配置刷入
> 
> 登录信息 192.168.1.1 注意不能和光猫冲突 root password

breed刷入后跟现在表现一样。wifi可用但是没网，有线连不上

NaiHeKK, 2022-07-12:
> > 用默认的配置去编译也不行吗
> 
> [kiddin9/OpenWrt_x86-r2s-r4s-r5s-N1](https://github.com/kiddin9/OpenWrt_x86-r2s-r4s-r5s-N1)这个仓库的应该是可以用的

这个仓库用的openwrt/openwrt源码，还不能改