Hyy2001X/AutoBuild-Actions #173: 更新了20220505版本后检查更新后出现问题了

ycbbcy, 2022-05-10:
当前固件版本: [36m[09:24:53][0m[31m 警告: 未检测到环境变量: Author[0m [36m[09:24:53][0m[31m 警告: 未检测到环境变量: Github[0m [36m[09:24:53][0m[31m 警告: 未检测到环境变量: TARGET_PROFILE[0m [36m[09:24:53][0m[31m 警告: 未检测到环境变量: TARGET_FLAG[0m [36m[09:24:53][0m[31m 警告: 未检测到环境变量: OP_VERSION[0m [36m[09:24:53][0m[31m 警告: 未检测到环境变量: OP_AUTHOR[0m [36m[09:24:53][0m[31m 警告: 未检测到环境变量: OP_BRANCH[0m [36m[09:24:53][0m[31m 警告: 未检测到环境变量: OP_REPO[0m 未知
云端固件版本: [36m[09:24:30][0m[31m 警告: 未检测到环境变量: Author[0m [36m[09:24:30][0m[31m 警告: 未检测到环境变量: Github[0m [36m[09:24:31][0m[31m 警告: 未检测到环境变量: TARGET_PROFILE[0m [36m[09:24:31][0m[31m 警告: 未检测到环境变量: TARGET_FLAG[0m [36m[09:24:31][0m[31m 警告: 未检测到环境变量: OP_VERSION[0m [36m[09:24:31][0m[31m 警告: 未检测到环境变量: OP_AUTHOR[0m [36m[09:24:31][0m[31m 警告: 未检测到环境变量: OP_BRANCH[0m [36m[09:24:31][0m[31m 警告: 未检测到环境变量: OP_REPO[0m

Hyy2001X, 2022-05-14:
试试执行 `tools` 并选择 `6. 环境修复`

lgfyxx, 2022-05-31:
我也是 打开 固件更新 也很慢，
环境修复不管用