Hyy2001X/AutoBuild-Actions #183: 2022年6月18日最新包更新后配置失效并且wifi功能不启用

MaxSecurity, 2022-06-19:
2022年6月18日最新包更新后配置失效并且wifi功能不启用，必须要插上网线才能修改配置和开启WiFi

Hyy2001X, 2022-06-19:
改成了闭源wifi驱动，建议不保留配置升级

MaxSecurity, 2022-06-19:
每次重启都要手动启动吗请问？

> 改成了闭源wifi驱动，建议不保留配置升级

Hyy2001X, 2022-06-19:
> 每次重启都要手动启动吗请问？
> 
> > 改成了闭源wifi驱动，建议不保留配置升级

不能下结论，手头上已经没有新三的机器了

MaxSecurity, 2022-06-19:
> > 每次重启都要手动启动吗请问？
> > > 改成了闭源wifi驱动，建议不保留配置升级
> 
> 不能下结论，手头上已经没有新三的机器了

目前解决的办法就是自启动中添加这两个命令
```
ifconfig ra0 up   #2G
ifconfig rai0 up  #5G
```

NaiHeKK, 2022-06-19:
> > 每次重启都要手动启动吗请问？
> > > 改成了闭源wifi驱动，建议不保留配置升级
> 
> 不能下结论，手头上已经没有新三的机器了

的确会这样，我已经重刷过的

Hyy2001X, 2022-06-19:
> > > 每次重启都要手动启动吗请问？
> > > > 改成了闭源wifi驱动，建议不保留配置升级
> > 
> > 
> > 不能下结论，手头上已经没有新三的机器了
> 
> 的确会这样，我已经重刷过的

好的，感谢反馈

MaxSecurity, 2022-06-19:
> > > > 每次重启都要手动启动吗请问？
> > > > > 改成了闭源wifi驱动，建议不保留配置升级
> > > 
> > > 
> > > 不能下结论，手头上已经没有新三的机器了
> > 
> > 
> > 的确会这样，我已经重刷过的
> 
> 好的，感谢反馈

我还发现了一个bug，就是WiFi的信号灯都不会亮了。。。。