Hyy2001X/AutoBuild-Actions #190: newifi-D2的R22.6.16-20220624固件5GHz wifi上传速度只有几十kb,退回R22.5.5-20220602正常.

nullnullnullnull0, 2022-07-02:
newifi-D2的R22.6.16-20220624固件5GHz wifi下载速度正常,但是上传速度只有几十kb.lan口直连的台式电脑上传速度没问题,确定是wifi问题.不保留配置,重新刷20220624固件也不行,最后退回R22.5.5-20220602正常.

ll3027, 2022-07-02:
遇到同样问题，感觉是turboacc的问题，换了新的硬件NAT加速，不启用都无法禁止，相关讨论在https://github.com/coolsnowwolf/lede/issues/9630 ，https://github.com/coolsnowwolf/lede/issues/9682以及https://github.com/coolsnowwolf/lede/issues/9450。

nullnullnullnull0, 2022-07-02:
> 遇到同样问题，感觉是turboacc的问题，换了新的硬件NAT加速，不启用都无法禁止，相关讨论在[coolsnowwolf/lede#9630](https://github.com/coolsnowwolf/lede/issues/9630) ，[coolsnowwolf/lede#9682](https://github.com/coolsnowwolf/lede/issues/9682) 以及 [coolsnowwolf/lede#9450](https://github.com/coolsnowwolf/lede/issues/9450) 。

@ll3027  老哥,2.5GHz wifi能ping通但是不能连接外网也是这个原因? @Hyy2001X 希望大佬能修复这个问题

Hyy2001X, 2022-07-03:
大家先退回 `R22.5.5-20220602` 吧，新版的固件撤了，等大佬修复

CyanidEEEEE, 2022-07-04:
很奇怪，我用0624给newifi d2一切正常，但是我自己编译的0701版本newifi d1的不行，这次的新hwnat看来问题有点随机

nullnullnullnull0, 2022-07-13:
20220712固件5g wifi正常了,但是2.5GHz wifi还是不能用,连接显示没有网络.