Hyy2001X/AutoBuild-Actions #202: Can add and change Wireless Interfaces

GenzPN, 2022-08-18:
Can new update code can be add Interfaces network like 
config interface 'usb'
        option proto 'dhcp'
        option ifname 'eth1'
and
SSID : Customname
Security : WPA2
Password : 1234567890

GenzPN, 2022-08-18:
[image]

Hyy2001X, 2022-08-19:
I cant understand your question, could you explain it with more detail plz

GenzPN, 2022-08-19:
Like the Sample Config Default

/etc/config/network

```
config interface 'loopback'
	option ifname 'lo'
	option proto 'static'
	option ipaddr '127.0.0.1'
	option netmask '255.0.0.0'

config globals 'globals'
	option packet_steering '1'
	option ula_prefix 'fd51:372c:8378::/48'

config interface 'lan'
	option type 'bridge'
	option ifname 'lan1 lan2'
	option proto 'static'
	option ipaddr '192.168.15.1'
	option netmask '255.255.255.0'
	option ip6assign '60'

config interface 'wan'
	option ifname 'wan'
	option proto 'dhcp'

config interface 'wan6'
	option ifname 'wan'
	option proto 'dhcpv6'
```
But i want add USB interfaces like this

```
config interface 'loopback'
	option ifname 'lo'
	option proto 'static'
	option ipaddr '127.0.0.1'
	option netmask '255.0.0.0'

config globals 'globals'
	option packet_steering '1'
	option ula_prefix 'fd51:372c:8378::/48'

config interface 'lan'
	option type 'bridge'
	option ifname 'lan1 lan2'
	option proto 'static'
	option ipaddr '192.168.15.1'
	option netmask '255.255.255.0'
	option ip6assign '60'

config interface 'wan'
	option ifname 'wan'
	option proto 'dhcp'

config interface 'wan6'
	option ifname 'wan'
	option proto 'dhcpv6'

config interface 'USB'
	option proto 'dhcp'
	option ifname 'eth1'
```

And the Wifi can custom 
/etc/config/wireless

```
config wifi-device 'radio0'
	option type 'mac80211'
	option path '1e140000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0'
	option channel '1'
	option band '2g'
	option htmode 'HT20'
	option disabled '0'
	option country 'VN'
	option legacy_rates '1'
	option mu_beamformer '1'

config wifi-iface 'default_radio0'
	option device 'radio0'
	option network 'lan'
	option mode 'ap'
	option encryption 'psk2'
	option key 'password'
	option ssid 'GenzPN 2.4G'

config wifi-device 'radio1'
	option type 'mac80211'
	option path '1e140000.pcie/pci0000:00/0000:00:01.0/0000:02:00.0'
	option channel '36'
	option band '5g'
	option htmode 'VHT80'
	option disabled '0'
	option country 'VN'
	option legacy_rates '1'
	option mu_beamformer '1'

config wifi-iface 'default_radio1'
	option device 'radio1'
	option network 'lan'
	option mode 'ap'
	option encryption 'psk2'
	option key 'password'
	option ssid 'GenzPN 5G'
```
It like if i want change everything in /etc/config it can overwriting