Hyy2001X/AutoBuild-Actions #208: How to set default language to English on building lede by sed  ?

AmirulAndalib, 2022-09-07:
Hi

I want to know how can I edit so that I can set default language to en_us by using SED on lede actions build of your repo ?

https://github.com/coolsnowwolf/lede

Thanks In Advance.

Hyy2001X, 2022-09-07:
[look this](https://github.com/coolsnowwolf/lede/issues/7942#issuecomment-930266103)

or add these commands 
> sed -i "s/zh_cn/en/g" package/lean/default-settings/files/zzz-default-settings
sed -i "s/+@LUCI_LANG_zh-cn/+@LUCI_LANG_en/g" package/lean/default-settings/Makefile
sed -i "/po2lmo ./po/zh-cn/default.po/d" package/lean/default-settings/Makefile

into Scripts/AutoBuild_DiyScript.sh -> Firmware_Diy() function

AmirulAndalib, 2022-09-07:
yes thank you thats what i was talking about

AmirulAndalib, 2022-09-07:
@Hyy2001X how to put luci theme neobird on `Scripts/AutoBuild_DiyScript.sh -> Firmware_Diy() function` ?

https://github.com/thinktip/luci-theme-neobird

Hyy2001X, 2022-09-07:
> @Hyy2001X how to put luci theme neobird on `Scripts/AutoBuild_DiyScript.sh -> Firmware_Diy() function` ?
> 
> https://github.com/thinktip/luci-theme-neobird

1. You can add command `git clone https://github.com/thinktip/luci-theme-neobird package/luci-theme-neobird` on Scripts/AutoBuild_DiyScript.sh -> Firmware_Diy() function

2. Don't forget  putting `CONFIG_PACKAGE_luci-theme-neobird=y` into your config

AmirulAndalib, 2022-09-07:
@Hyy2001X please add support for xiaomi mi router 4A gigabit Edition 1200AC MTK_7621

Hyy2001X, 2022-09-07:
> @Hyy2001X please add support for xiaomi mi router 4A gigabit Edition 1200AC MTK_7621

Thanks for your suggestion, but I am sorry that I have no such device to test, so its difficult to support this device

AmirulAndalib, 2022-09-07:
I can try

AmirulAndalib, 2022-09-07:
I have the router an some example repository

AmirulAndalib, 2022-09-07:
> 1. You can add command `git clone https://github.com/thinktip/luci-theme-neobird package/luci-theme-neobird` on Scripts/AutoBuild_DiyScript.sh -> Firmware_Diy() function
> 2. Don't forget  putting [...]

Also how to set neobird theme as default using sed ?

Hyy2001X, 2022-09-09:
> I can try

Okay, you can download firmware at https://github.com/Hyy2001X/AutoBuild-Actions/releases/download/AutoUpdate/AutoBuild-lede-xiaomi_mi-router-4a-gigabit-R22.9.1-20220909-Full-ba44d.bin

AmirulAndalib, 2022-09-10:
So apparently I am using Openwrt 22.03 So can I upgrade it using the sysupgrade file and keep netowkr setting backed up then put it back after flashing Lede ?

Hyy2001X, 2022-09-10:
> So apparently I am using Openwrt 22.03 So can I upgrade it using the sysupgrade file and keep netowkr setting backed up then put it back after flashing Lede ?

lede-luci version is still at 18.06, this operation may cause incompatible

AmirulAndalib, 2022-09-16:
Ok

AmirulAndalib, 2022-10-09:
@Hyy2001X for compiling 1200ac firmware for Xiaomi R4A gigabit you can take ideas and diy patches from here

https://github.com/ababwbq/Actions-OpenWrt-Xiaomi-R4A