Hyy2001X/AutoBuild-Actions #210: 我直接fork了整个项目，但是我用actions给新三编译时，在build openwrt阶段遇到错误

IDeLoveYou, 2022-09-12:
我尝试了三四次都以失败告终，大佬能否提示一下
[image]

IDeLoveYou, 2022-09-13:
[logs_1.zip](https://github.com/Hyy2001X/AutoBuild-Actions/files/9552779/logs_1.zip)
大佬，这是acaction下载的完整日志

Hyy2001X, 2022-09-13:
再重新编译一下看看

IDeLoveYou, 2022-09-13:
我从前天编译到现在，试了五六次，上面那个日志是昨天晚上跑的，今天早上又报错了，我在想是不是我的步骤出了问题，但是我就编译了newifi的lite版，应该不会错

Hyy2001X, 2022-09-13:
> 我从前天编译到现在，试了五六次，上面那个日志是昨天晚上跑的，今天早上又报错了，我在想是不是我的步骤出了问题，但是我就编译了newifi的lite版，应该不会错

步骤没问题，我看这个问题已经在10小时前被修复了 https://github.com/coolsnowwolf/lede/commit/64e592e2d1abbad69dce56143ba5186f363015c2

IDeLoveYou, 2022-09-13:
是嘛，哇哇哇，被坑到了

IDeLoveYou, 2022-09-13:
大佬，我想问一下，新三的闭源wifi驱动目前是有什么已知bug嘛，我看你的config里用的是开源驱动

Hyy2001X, 2022-09-13:
> 大佬，我想问一下，新三的闭源wifi驱动目前是有什么已知bug嘛，我看你的config里用的是开源驱动

闭源用着比较膈应，而且现在的开源驱动速度也不输闭源了

IDeLoveYou, 2022-09-13:
got it，我现在来编译试一下，看看能不能过

IDeLoveYou, 2022-09-13:
还是过不了
[logs_4.zip](https://github.com/Hyy2001X/AutoBuild-Actions/files/9553692/logs_4.zip)

wangcome, 2022-09-13:
[image]
me too

IDeLoveYou, 2022-09-13:
好像修复了，我还没开始编译
https://github.com/coolsnowwolf/lede/issues/10091#issuecomment-1245029245

IDeLoveYou, 2022-09-14:
亲测。config文件里把CONFIG_PACKAGE_luci-app-samba=y注释了就行了，要不然编译不过

closed 2022-09-14 by IDeLoveYou

lgfyxx1, 2022-09-19:
同错误，可是想要samba共享怎么办

Hyy2001X, 2022-09-19:
> CONFIG_PACKAGE_luci-app-samba=y

注释 CONFIG_PACKAGE_luci-app-samba=y
选上 CONFIG_PACKAGE_autosamba=y